zephyrproject-rtos/zephyr #58222: CI: kernel: userspace: kernel.memory_protection.userspace: mimxrt595/685: mpu part test fail

hakehuang, 2023-05-24:
This issue list is only for bugs in the main Zephyr code base

**Describe the bug**

```
SUITE FAIL -  50.00% [userspace_domain]: pass = 2, fail = 2, skip = 0, total = 4 duration = 0.095 seconds
- PASS - [userspace_domain.test_1st_init_and_access_other_memdomain] duration = 0.040 seconds
- FAIL - [userspace_domain.test_domain_add_part_drop_to_user] duration = 0.027 seconds
- PASS - [userspace_domain.test_domain_add_thread_drop_to_user] duration = 0.002 seconds
- FAIL - [userspace_domain.test_domain_remove_part_drop_to_user] duration = 0.026 seconds
SUITE FAIL -  33.33% [userspace_domain_ctx]: pass = 1, fail = 2, skip = 0, total = 3 duration = 0.056 seconds
- FAIL - [userspace_domain_ctx.test_domain_add_part_context_switch] duration = 0.027 seconds
- PASS - [userspace_domain_ctx.test_domain_add_thread_context_switch] duration = 0.001 seconds
- FAIL - [userspace_domain_ctx.test_domain_remove_part_context_switch] duration = 0.028 seconds
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 mimxrt595_cm33_evk / mimxrt685_cm33_evk
 - What have you tried to diagnose or workaround this issue?
 MPU configuration for rtxxx soc are limited
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
NO

**To Reproduce**
Steps to reproduce the behavior:
> cd tests/kernel/mem_protect/userspace/
> west build -b mimxrt595_evk_cm33 -- -DCONFIG_TEST_HW_STACK_PROTECTION=n -DCONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=0
> west flash

**Expected behavior**
Test PASS

**Impact**
userspace support

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.3.0-4123-gd01780fc9403 ***
Running TESTSUITE userspace
===================================================================
START - test_access_after_revoke
E: thread 0x30180fb0 (1) does not have permission on k_sem 0x30183ad8
E: permission bitmap
E: 04 00                   |..      
E: syscall z_vrfy_k_sem_take failed check: access denied
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0xeffdfffb
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x30180fb0 (test_access_after_revoke)
Caught system error -- reason 3
System error was expected
PASS - test_access_after_revoke in 0.050 seconds
===================================================================
START - test_bad_syscall
E: Bad system call id 2147483647 invoked
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0xeffdfffb
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x30180fb0 (test_bad_syscall)
Caught system error -- reason 3
System error was expected
PASS - test_bad_syscall in 0.036 seconds
[... 405 lines cut ...]
- PASS - [userspace.test_write_kernram] duration = 0.041 seconds
- PASS - [userspace.test_write_kernro] duration = 0.040 seconds
- PASS - [userspace.test_write_kerntext] duration = 0.041 seconds
- PASS - [userspace.test_write_kobject_user_pipe] duration = 0.042 seconds
- PASS - [userspace.test_write_other_stack] duration = 0.040 seconds
- PASS - [userspace.test_write_priv_stack] duration = 0.041 seconds
SUITE FAIL -  50.00% [userspace_domain]: pass = 2, fail = 2, skip = 0, total = 4 duration = 0.095 seconds
- PASS - [userspace_domain.test_1st_init_and_access_other_memdomain] duration = 0.040 seconds
- FAIL - [userspace_domain.test_domain_add_part_drop_to_user] duration = 0.027 seconds
- PASS - [userspace_domain.test_domain_add_thread_drop_to_user] duration = 0.002 seconds
- FAIL - [userspace_domain.test_domain_remove_part_drop_to_user] duration = 0.026 seconds
SUITE FAIL -  33.33% [userspace_domain_ctx]: pass = 1, fail = 2, skip = 0, total = 3 duration = 0.056 seconds
- FAIL - [userspace_domain_ctx.test_domain_add_part_context_switch] duration = 0.027 seconds
- PASS - [userspace_domain_ctx.test_domain_add_thread_context_switch] duration = 0.001 seconds
- FAIL - [userspace_domain_ctx.test_domain_remove_part_context_switch] duration = 0.028 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: f8b1ec0de6fbed7c927e65d76b4897b4
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, , )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used:  zephyr-v3.3.0-4123-gd01780fc9403

hakehuang, 2023-05-24:
@dleach02 , this MPU configuration in RTxxx series seems limited the test, shall I add a PR to skip those test?

decsny, 2023-05-31:
Hi @hakehuang, I looked into this issue and I think this test should not be included as it is highlighting that we are not being careful with how many mpu regions we define.

In the case of this failure, the situation is this:
The RT595 has 8 mpu regions.

We are currently defining 5 static regions:
Flash,
SRAM0 (main ram region)
SRAM1 (usb ram region)
RAMFUNC (all arm platforms define this)
Flexspi code relocation region 

That leaves us with 3 mpu regions available for dynamic usage. This test requires 4:
User thread stack region,
Ztest region
2 regions defined by the test, "default" and "alt"

I think this is therefore not a "bug" but an unoptimized implementation. For example, I have a suspicion that the flexspi code can be put in the ramfunc region which seems to have the same purpose of relocating text to ram as the custom one we defined by adding a new kconfig `CONFIG_CODE_DATA_RELOCATION_SRAM`. Also, we probably do not need to define a USB RAM MPU region if USB is not being used in the build. These two optimizations are enhancements that might allow a test such as this one to pass without running into the issue of having no available MPU regions.

decsny, 2023-06-01:
For now since this is the bug fix period of the release cycle, I am just disabling the USB RAM for that test in the PR I just linked to this issue. There is not really a bug here and any other "fixes" to this issue will require a lot of changes that I am not comfortable justifying as a bug fix. We still want to support this test as it is importantly testing memory protection and userspace so I don't think excluding it is appropriate, either. So I think adding this DT board overlay to the test so that it passes is fair to close this issue. I also added a comment to the board DTS to make it more clear to customers that this is possible.

PR #58754 (2023-06-01): tests: userspace: Support RT 3 digit
RT 3 digit ran out of mpu regions to do this test, so add board overlays to disable the USB SRAM MPU region definition because it is not needed for this test.

Also add a comment to the board dts making it clear that this is possible.

Fixes #58222