zephyrproject-rtos/zephyr #58297: Device Initialization priority based on DTS

najumon1980, 2023-05-26:
**Issue/problem Statement.**
======================================

Current Zephyr device model provide developers to define their driver Initialization order and priority using Initialization level and priority with in each level. This satisfy platform with small number of IO interface and doesn't have more complex parent-child device hagiarchy. But it will be difficult to manage if platform have more number of IOs and their instances,  have child node etc. Following are the list of few issues that can be high lighted:

- There is no guarantee that parent driver instantiated before its child node device init invoked. 
- Configure device Initialization priority in a centralized manner: An application developer might find it difficult to define the device dependency and Initialization priority. He might need to walk through driver details and play around with  initialization level and priority of each drivers to meet his requirements.
- It is difficult to configure Initialization level and priority per instance.

Some of the above issues can be avoided or minimized If we enhance existing kernel device initialization sequence in such a way that kernel will enumerate and initialize devices based on their node position in the device tree source (DTS) and parent-child node relationship. 

> **Note**
Please note that the PR is only a draft version and currently given for analyze purpose (will refine/optimize it once aligned on requirements/design). It adopt and modify existing elf_parser.py and gen_handles.py post build script to create device Initialization priority list.

Issues with similar problem statements and suggestions:
[Device initialization order that respects devicetree dependencies ](https://github.com/zephyrproject-rtos/zephyr/issues/22545)

**How to enhance interface for IO drivers - Draft Proposal**
======================================
Following section brief on the new proposal for enhance kernel device initialization flow. The proposal is only able to capture base requirements, but might need continue improvement by review and adopt suggestion from community and domain experts. 

**Analyze on Architecture Changes on Existing driver Interface**
------------------------------------------------------------------

In the proposed model, driver initialization level limited to pre-kernel, post-kernel or application level which are based on the specific driver implementation ( example: if driver have to use some kernel APIs during init then it should be post kernel etc.) and use case (example: UART driver might required to be pre-kernel for enable early logging). The Initialization priority is completely depend on the position of nodes in the device tree and  also depend on bus/parent node Initialization priority. The Initialization level expected to be remain same for a specific driver and will not change from platform to platform. All other inter driver dependency and Initialization priority will be take care based on device/node position in the device tree such as order of device node, parent node up on which the child node located etc. The kernel will ensure the parent node init will be invoked before the child node init call. Similarly device init will be called in the ascending order of nodes in the device tree. This will help a platform integrator or application writer to configure the driver Initialization order dependency based on his platform device initialization order requirement, bus upon which the device node placed etc. Following diagram depict the parent child hierarchy and the device initialization order. Please note that below diagram assume all driver have same initialization priority say post-kernel and if it it have different initialization level then all pre-kernel device will get instantiated first followed by post-kernel and finally application level.
 
[image]

**Parent-child device/node dependency**

For device node such as PCI host, SPI, I2C etc. might have child device/node and it is expected that parent device should 1st instantiate before child device init invoked. In the proposed model,  Zephyr post build script will recursively walk through the device tree and elf file to collect and sort all the device initialization function based on their node position and parent-child relationship. 

**Configure device initialization priority in a centralized manner**

Developers shall able to configure device initialization order/dependency from a centralized manner and without knowing much details of driver complexity.  For example, in following device tree,  there are two parent node under root node. So the kernel will enumerate parent "a device" and then all it child devices (child "a" to "c" device) then go to parent "b device" node and so on. This also depend on initialization level of node as well as it parent node/nodes. Developer can easily change position of device node based on their initialization order requirements via device tree.

```diff

{
   parent_dev_a: parent_dev_a0 {
     ....
	  child_dev_a: child_dev_a0 {
                ...
	  };
	  child_dev_b: child_dev_b0 {
                ...
	  };
	  child_dev_c: child_dev_c0 {
                ...
	  };
  }
 parent_dev_b: parent_dev_b0 {
  ...
   };
};
```

> **Note**
Kernel will enumerate each child device and invoke their driver init function in the assenting order of their position in device tree. Device drivers can use existing DT_INST_CHILD() or DT_INST_PARENT() macros to obtain child or parent nodes reference. Usage of device ordinal APIs such as device_required_foreach() or device_supported_foreach() can be still possible for finding additional device dependency, but usage might reduce with the new interface.

**Configure initialization priority per instance**

Developers shall able to configure different instances of same driver with different initialization priority. Since initialization priority is depend up node position in DTS, developers can easily control priority via changing node position in the device tree.

> **Note**
Initialization level still need to provide as part of DEVICE_DT_DEFINE macro since it specific to a type of devices or diver. For example some driver need to use scheduler/kernel API and hence should be initialization only after post kernel. Similarly some drivers need be instantiate before kernel and these drivers can be always defined as pre-kernel level. Please note that all pre-kernel drivers will instantiate 1st followed by post kernel and then finally application level. But order of initialization in each level will be depend on node position in the DTS.

**Co-exist with non DTS based drivers and backward compatibility**

Developer shall able to use non DTS based drivers which are defined using DEVICE_DEFINE() along with DTS based drivers defined using DEVICE_DT_DEFINE(). Current proposed model and the draft PR provided is taken care of this and allow to co-exist both drivers, but non DTS based drivers initialization priority cannot be controlled via DTS.  For a given initialization level, DTS based drivers will be instantiated first followed by non DTS based drivers. Developer can enable the new features using CONFIG_DTS_BOOT_PRIORITY config macro and if enabled it will change the initialization priority of all existing drivers which defined using DEVICE_DT_DEFINE().  CONFIG_DTS_BOOT_PRIORITY by default disabled so that application writer can enable only if they require the new model.

> **Note**
Please note that in the new model, prio argument in the DEVICE_DT_DEFINE() macro will be don't care and the priority will be decided by node position in the DTS.

**Device Power Management**
The Zephyr device power management subsystem can make use of the new sequence for suspend and resume the device in a centralize manner since it can call the suspend/resume API without worry of driver's dependency. A node which initialized first should be suspended last and also parent device should be remain ON until all its child device got suspended. Similarly resume should follow exactly same order that of initialization sequence during boot flow. 

**Performance impact**

There shall be no boot latency and least code blot because of introduce the new DTS based boot model. The proposed model use post-build script to generate device list with initialization level based on driver definition and initialization priority based on node position in the DTS. It also sue separate linker section and hence will not be included in default init.

Following table given as an example which summarize flash and SRAM memory usage with and without CONFIG_DTS_BOOT_PRIORITY enabled: The binary build for sam4l_ek platform/board.

| DTS base Boot | FLASH | SRAM |
| ---------------- |---------|--------|
|  YES                  | 29699  |  5568  |
|  NO                   | 29163 |  5504  |

Increase in SRAM usage is expected to be NIL and the extra 64byte increase need be analyze for the draft PR.

**PR details**

gmarull, 2023-05-26:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/22545 or https://github.com/zephyrproject-rtos/zephyr/issues/34518

tbursztyka, 2024-02-20:
It's a known issue from the start with existing opened issues as @gmarull noted, no need to discuss about the same in various places. Closing.

closed 2024-02-20 by tbursztyka