zephyrproject-rtos/zephyr #58303: building failure with picolibc

FRASTM, 2023-05-26:
**Describe the bug**
When building the tests with PICOLIBC there is a compilation error on several (all ?) stm32 boards occurs:
`FAILED: zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj`
`arm-zephyr-eabi-gcc: fatal error: cannot read spec file 'picolibc.specs': No such file or directory`
`compilation terminated.`

 seen with :
- samples/basic/hash_map/libraries.hash_map.picolibc.separate_chaining.djb2
- tests/kernel/common/kernel.common.picolibc
- tests/lib/c_lib/libraries.libc.picolibc,
- tests/lib/cbprintf_fp/libraries.cbprintf_fp.printf.picolibc
- tests/lib/cbprintf_fp/libraries.cbprintf_fp.printfcb.picolibc
- tests/lib/sprintf/libraries.libc.picolibc.sprintf
- tests/lib/sprintf/libraries.libc.picolibc.sprintf_new
- tests/misc/print_format/printk.format.picolibc
- tests/posix/common/portability.posix.common.picolibc
- tests/posix/eventfd/portability.posix.eventfd.picolibc
- tests/posix/eventfd_basic/portability.posix.eventfd_basic
- tests/posix/eventfd_basic/portability.posix.eventfd_basic.picolibc.posix_api,
- tests/posix/getopt/portability.posix.getopt.picolibc
- tests/posix/headers/portability.posix.headers.picolibc.with_posix_api
- tests/posix/headers/portability.posix.headers.picolibc.without_posix_api

Please also mention any information which could help others to understand
the problem you're facing:
 -  looking at commit  2200861b96c3ef6fd76a49a3ccc9a6ad492944e4
 

**Expected behavior**
compilation completes successfully

**Impact**
CI failure 

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain : sdk-0.15.2 or sdk-0.15.1
 - Commit SHA  : 37c2a0d4a6f8d7c4c29c3a60a45407747d28c2c8

stephanosio, 2023-05-26:
@FRASTM Please upgrade your Zephyr SDK version to v0.16.0 or above (v0.16.1 is the latest at the time of writing).

FRASTM, 2023-05-30:
fixed with sdk 0.16.1

closed 2023-05-30 by FRASTM