zephyrproject-rtos/zephyr #58305: New host simulation target: Embedded simulator

aescolar, 2023-05-26:
## Intro / What it is this

* A new target board (emsim_sc), which is an evolution of native_posix.
* You still end up with a native Linux executable you can debug and instrument like with native_posix. You can build with sanitizers, gprof, get coverage reports, etc.
* Supports (optionally) building Zephyr with an embedded libC. With this option there is no conflicts with the host libC anymore.
* Still supports building with the host libC for when interacting with the host with ease is the priority.
* Supports AMP builds (with upcoming target boards). That is, for ex., 2 simulated µC, each with their own Zephyr instance, app etc.

## What is *not* changing

* The POSIX architecture, native_posix and nrf52_bsim remain
* All boards based on the POSIX arch (in and off-tree) keep working.

## Problem this is solving

* There is 3 things we aim at solving:
    - Currently we cannot support AMP like simulations with the POSIX architecture. AMP SOCs have become very normal, and with it the need to be able to test in workstation the interaction between the SW in different µC, between different OSs and/or with baremetal SW / controllers running in the same SOC.
    - native_posix (and the old POSIX architecture boards) only allow building with the host libC. This is simple, but limits what can be built due to header/symbol name collisions.

## How it works in short / how it differs from what we had

* Before the Zephyr OS, drivers, application, the POSIX arch, and the "board"(runner/scheduler/hw models) code were built using the host libC, and linked in one step. In some cases this causes symbols conflicts.
* With this new target:
    - Zephyr OS, drivers, application, and the top of the arch are built first.
      With a choice of the host libC or an embedded libC.
    - The "runner", and the parts of the architecture and "HW models"/drivers bottoms that require building with the Linux OS are built in a separate step.
    - Both are linked together into 1 executable.
    - The runner (embedded simulator) is Zephyr agnostic: It can be used to run Zephyr, but it can also be used to run other embedded OSes, or baremetal embedded apps.

* Now you can chose which libC you use: host, or one of the embedded ones.
    - When building the embedded part with the host libC you have the same name/symbol conflicts limitation as before.
    - When building with an embedded libC, interaction with the host can only be done (easily) using the runner as a gateway into the host. That is, native drivers need to be split into an "embedded part" build with Zephyr, and a "bottom"/"hw model" part built with the runner.

## What is the status:

* At this point I have a prototype with all twister tests that target native_posix passing, all POSIX compatibility API shim tests passing, and tests for libC for minimal libC and picolibC (module) passing (the ones I tried).
* For the embedded part built with the embedded libC there is so far only a console driver.
* I'm polishing it to merge-able condition.

## Proposed way forward:

* The embedded simulator with a first Zephyr target board (emsim_sc, embedded simulator, single core) are contributed to Zephyr.
* While they are matured native_posix remains the primary native target test platform.
* Once the emsim_sc has proven reliable, we switch CI to use it as primary target.
* If there is no issues after a few weeks, native_posix (the board) is removed, and a compatibility mode is left in emsim_sc, so apps/drivers that targeted native_posix can be run unchanged on emsim_sc.
* In the meanwhile, more drivers for the emsim_sc are added which can benefit of the host OS while being built with the embedded libC (refactored versions of the native_posix ones).

* Notes:
    - This does not affect other posix arch boards.
    - You can expect new AMP boards being introduced in the future based on this same embedded simulator, emulating real HW (akin to the nr52_bsim).

## Relevant choices so far

* Name of the runner (embedded simulator) and target board (emsim single core). They are not settled.
* This new board keeps using the POSIX architecture and its "SOC", instead of adding a new dedicated architecture:
    - A new dedicated architecture:
        - Would allow us to retire the POSIX arch (and therefore fix the "posix" name confusion).
        - But would duplicate a lot of code, docs, require quite a lot of extra minor changes in-tree, and leave the old boards (specially out of tree) in a limbo state.
    - By keeping the same POSIX arch there is some minor refactoring of the POSIX arch and SOC code (into a top and bottom) to enable using it in the embedded simulator. (no API or functional changes, just splitting some code in separate files), and docs updates.
    - I lean strongly to maintain backwards compatibility with the old architecture and old boards.
* native_posix remains as is while emsim_sc matures:
    - It would be possible to just have native_posix be a compatibility shim/mode over emsim_sc from day 1.
    - But, native_posix is used very heavily and it is very reliable today. Basing it on emsic_sc introduces the risk of regressions. As emsim_sc supports new use cases, I expect very active development at the beginning, and therefore possible regressions in the first weeks/months.
    - CI reliability would outweigh the benefits of early cleanup.

## A drawing of the architecture
[image]

aescolar, 2023-05-26:
Fixes #13054 , #45100 , #47856

carlescufi, 2023-06-13:
Architecture WG:

- Lively discussion around the details already described in the RFC text
- The POSIX architecture will remain, but the `native_posix` board will slowly be deprecated and replaced by `natsim_sc`
- @nashif asks about the timeline: @aescolar states that a PR can follow immediately, then have a single release where natsim is experimental, and then after deprecate native_posix and keep it only for 2 releases. After that, the native_posix compatibility mode in natsim can be used
- @carlescufi asks about the multi-stage linking implications with this new model. @aescolar answers that each instance can be built and pre-linked as many times as necessary

sjg20, 2023-06-13:
Just to check, Zephyr is still running as a single Linux process, right?

aescolar, 2023-06-14:
> Just to check, Zephyr is still running as a single Linux process, right?

@sjg20 Yes. At runtime it looks practically the same as with native_posix.

aescolar, 2023-07-05:
Closed as done.

closed 2023-07-05 by aescolar