zephyrproject-rtos/zephyr #58310: ESP32C3 MCUBOOT Error

ShahinHaque97-Corintech, 2023-05-26:
Hello, 

As of yesterday 25/05/2023 ESP32 (All supported variants ESP32, ESP32C3 etc etc ) directory and docs have been updated to support MCUBOOT rather than ESP-IDF

https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/riscv/esp32c3_devkitm/doc/index.rst
https://docs.zephyrproject.org/latest/boards/riscv/esp32c3_devkitm/doc/index.html

With a fresh west init of zephyr, 

Regular build works as intended
`west build -b esp32c3_devkitm samples/hello_world `

Log
```
[181/181] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          28 B         32 B     87.50%
             ROM:       23534 B    4194240 B      0.56%
     iram0_0_seg:        8016 B       320 KB      2.45%
     irom0_0_seg:       78262 B    4194272 B      1.87%
     drom0_0_seg:        2040 B    4194240 B      0.05%
     dram0_0_seg:       17924 B       320 KB      5.47%
    rtc_iram_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
esptool.py v4.5
Creating esp32c3 image...
Merged 6 ELF sections
Successfully created esp32c3 image.
```

However, MCUBOOT fails
`west build -b esp32c3_devkitm --sysbuild samples/hello_world  -p`

Log
```
[155/165] Linking C executable zephyr\zephyr_pre0.elf

[159/165] Linking C executable zephyr\zephyr_pre1.elf

[165/165] Linking C executable zephyr\zephyr.elf
FAILED: zephyr/zephyr.elf zephyr/zephyr.map zephyr/zephyr.bin zephyr/zephyr.lst zephyr/zephyr.stat zephyr/zephyr.signed.bin zephyr/zephyr.signed.confirmed.bin C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.map C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.bin C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.lst C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.stat C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.signed.bin C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.signed.confirmed.bin
cmd.exe /C "cd . && C:\zephyr-sdk-0.16.1-beta1\riscv64-zephyr-elf\bin\riscv64-zephyr-elf-gcc.exe  -gdwarf-4 zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj zephyr/CMakeFiles/zephyr_final.dir/dev_handles.c.obj zephyr/CMakeFiles/zephyr_final.dir/isr_tables.c.obj -o zephyr\zephyr.elf  zephyr/CMakeFiles/offsets.dir/./arch/riscv/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker.cmd  -Wl,-Map=C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr_final.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/riscv/core/libarch__riscv__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:/zephyr-sdk-0.16.1-beta1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0"  -LC:/zephyr_esp_temp/zephyr/build/hello_world/zephyr  -lgcc  -Wl,--print-memory-usage  zephyr/arch/common/libisr_tables.a  -lgcc  -mabi=ilp32  -march=rv32imc_zicsr  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -TC:/zephyr_esp_temp/modules/hal/espressif/zephyr/esp32c3/src/linker/esp32c3.rom.alias.ld  -TC:/zephyr_esp_temp/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.ld  -TC:/zephyr_esp_temp/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld  -TC:/zephyr_esp_temp/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld  -TC:/zephyr_esp_temp/modules/hal/espressif/zephyr/esp32c3/../../components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld  -TC:/zephyr_esp_temp/modules/hal/espressif/zephyr/esp32c3/../../components/soc/esp32c3/ld/esp32c3.peripherals.ld && cmd.exe /C "cd /D C:\zephyr_esp_temp\zephyr\build\hello_world\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E copy zephyr_final.map zephyr.map && C:\zephyr-sdk-0.16.1-beta1\riscv64-zephyr-elf\bin\riscv64-zephyr-elf-objcopy.exe --gap-fill 0xff --output-target=binary --remove-section=.comment --remove-section=COMMON --remove-section=.eh_frame zephyr.elf zephyr.bin && C:\zephyr-sdk-0.16.1-beta1\riscv64-zephyr-elf\bin\riscv64-zephyr-elf-objdump.exe -d -S zephyr.elf > zephyr.lst && C:\zephyr-sdk-0.16.1-beta1\riscv64-zephyr-elf\bin\riscv64-zephyr-elf-readelf.exe -e zephyr.elf > zephyr.stat && C:\Python311\python.exe -m west sign --quiet --tool imgtool --tool-path C:/Users/Shahi/AppData/Roaming/Python/Python311/Scripts/imgtool.exe --build-dir C:/zephyr_esp_temp/zephyr/build/hello_world --bin --sbin C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.signed.bin -- && C:\Python311\python.exe -m west sign --quiet --tool imgtool --tool-path C:/Users/Shahi/AppData/Roaming/Python/Python311/Scripts/imgtool.exe --build-dir C:/zephyr_esp_temp/zephyr/build/hello_world --bin --sbin C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.signed.confirmed.bin -- --pad --confirm""
Memory region         Used Size  Region Size  %age Used
     mcuboot_hdr:          32 B         32 B    100.00%
        metadata:          32 B         32 B    100.00%
             ROM:       78318 B    4194240 B      1.87%
     iram0_0_seg:        8352 B       320 KB      2.55%
     irom0_0_seg:      143918 B         4 MB      3.43%
     drom0_0_seg:        2204 B    4194240 B      0.05%
     dram0_0_seg:       18292 B       320 KB      5.58%
    rtc_iram_seg:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
FATAL ERROR: command exited with status 1: C:/Users/Shahi/AppData/Roaming/Python/Python311/Scripts/imgtool.exe sign --version 0.0.0+0 --align 4 --header-size 32 --slot-size 1048576 'C:\zephyr_esp_temp\zephyr\build\hello_world\zephyr\zephyr.bin' C:/zephyr_esp_temp/zephyr/build/hello_world/zephyr/zephyr.signed.bin
ninja: build stopped: subcommand failed.

[10/16] No configure step for 'mcuboot'
FAILED: hello_world-prefix/src/hello_world-stamp/hello_world-build C:/zephyr_esp_temp/zephyr/build/hello_world-prefix/src/hello_world-stamp/hello_world-build
cmd.exe /C "cd /D C:\zephyr_esp_temp\zephyr\build\hello_world && "C:\Program Files\CMake\bin\cmake.exe" --build ."
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build C:/zephyr_esp_temp/zephyr/build
```

To reproduce

Follow the docs

`west init zephyr_esp`
`west build -b esp32c3_devkitm --sysbuild samples/hello_world  -p`

Add CONFIG_BOOTLOADER_MCUBOOT=y to the hello_word proj file

marekmatej, 2023-05-26:
@ShahinHaque97-Corintech Thanks for the report! Can you make sure you did update hal: `west update hal_espressif`? If that won't help please make sure your feature branch you might using is rebased.

ShahinHaque97-Corintech, 2023-05-26:
@marekmatej 

`west update hal_espressif` does not solve this, the branch I am using is the main branch (new west init), I am compared the ESP32C3 files in the board folder and can confirm this is the main branch

marekmatej, 2023-05-26:
> Add CONFIG_BOOTLOADER_MCUBOOT=y to the hello_word proj file

Could you please make sure you also updated the mcuboot: `west update mcuboot` ?
Also, check that `bootloader/mcuboot` and `modules/hal/espressif` are on the commits provided by the `west.yml` in the Zephyrs main branch.

Also, note that if using `--sysbuild` there is no need to enable `CONFIG_BOOTLOADER_MCUBOOT` by hand as it is done in the sysbuild Kconfig.

ShahinHaque97-Corintech, 2023-05-26:
Unfortunately I have already tried everything suggested. `west update mcuboot` , changing the yaml to the latest of each repository. 
 --sysbuild without CONFIG_BOOTLOADER_MCUBOOT and regular build with.

ShahinHaque97-Corintech, 2023-05-26:
Edit, they were on the same as the main branch already  and I have tested latest commit for mcuboot & espressif

sylvioalves, 2023-05-26:
@ShahinHaque97-Corintech, I've tried and it worked properly. Windows 10, sdk 0.16.1. Will post some log output later.

ShahinHaque97-Corintech, 2023-05-30:
[espressif_mcuboot_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/11597213/espressif_mcuboot_log.txt)

Here is my attached full log:
Win10, SKD 0.61.1

ShahinHaque97-CT, 2023-06-01:
@sylvioalves Have you had a chance to look at the logs?

sylvioalves, 2023-06-01:
@ShahinHaque97-CT, what is the imgtool version you have been using? Also, have you run `pip3 install -r scripts/requirements.txt`?

closed 2023-08-02 by ShahinHaque97-CT