zephyrproject-rtos/zephyr #58316: <ws-path>/zephyr/drivers/i2c/i2c_gpio.c:144:54: warning: passing argument 3 of 'i2c_bitbang_init' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]

IvanVnucec, 2023-05-26:
**Describe the bug**
`<ws-path>/zephyr/drivers/i2c/i2c_gpio.c:144:54: warning: passing argument 3 of 'i2c_bitbang_init' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]` warning when `CONFIG_I2C_GPIO=y`.

 - using nRF52832
 - using si7210 sensor with I2C GPIO driver.

**To Reproduce**
Add this overlay and Kconfigs:
```dts
/ {
	gpio_i2c0 {
		compatible = "gpio-i2c";
		status = "okay";
		clock-frequency = <I2C_BITRATE_STANDARD>;
		sda-gpios = <&gpio0 18 (GPIO_OPEN_DRAIN)>;
		scl-gpios = <&gpio0 17 (GPIO_OPEN_DRAIN)>;
		label = "GPIO_I2C_0";
		#address-cells = <1>;
		#size-cells = <0>;

		si7210:si7210@30 {
			compatible = "silabs,si7210";
			reg = <0x30>;
			label = "Magnetic proximity sensor";
			status = "okay";
		};
	};
};
```
```kconfig
CONFIG_I2C=y
CONFIG_I2C_GPIO=y
CONFIG_SENSOR=y
CONFIG_PM_DEVICE=y
CONFIG_SI7210=y
```

Import .overlay snippets mentioned above and enable Kconfig flags mentioned above.  
Also, I have all warnings enabled in my app build.

**Expected behavior**
The warning should not occur.

**Impact**
The impact is the clutter of the build. Also, the kernel shouldn't have any warnings.

**Logs and console output**
Here is the last snippet of the build log:
```cmd
-- Zephyr version: 3.2.99 (<ws-path>/zephyr), build: v3.2.99-ncs2
[163/251] Building C object zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_gpio.c.obj
<ws-path>/zephyr/drivers/i2c/i2c_gpio.c: In function 'i2c_gpio_init':
<ws-path>/zephyr/drivers/i2c/i2c_gpio.c:144:54: warning: passing argument 3 of 'i2c_bitbang_init' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  144 |         i2c_bitbang_init(&context->bitbang, &io_fns, config);
      |                                                      ^~~~~~
In file included from <ws-path>/zephyr/drivers/i2c/i2c_gpio.c:38:
<ws-path>/zephyr/drivers/i2c/i2c_bitbang.h:43:64: note: expected 'void *' but argument is of type 'const struct i2c_gpio_config *'
   43 |                         const struct i2c_bitbang_io *io, void *io_context);
      |                                                          ~~~~~~^~~~~~~~~~
[241/251] Linking C executable zephyr\zephyr_pre0.elf

[245/251] Linking C executable zephyr\zephyr_pre1.elf

[251/251] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      189680 B       512 KB     36.18%
             RAM:       35257 B        64 KB     53.80%
        IDT_LIST:          0 GB         2 KB      0.00%
```

**Environment (please complete the following information):**
 - OS: Windows 10
 - Zephyr SDK
 - Zephyr version v3.2.99-ncs2

nashif, 2023-05-26:
Please keep the title brief and do not put warning/error output in the title. Describe what the issue instead.

carlescufi, 2023-05-29:
This was fixed by https://github.com/zephyrproject-rtos/zephyr/commit/aced3bb2ac1e2e69a35f74edb723a08f57f33aab. Thanks for reporting!

cc @gmarull

closed 2023-05-29 by carlescufi