zephyrproject-rtos/zephyr #58336: qdec_nrfx regression after nrfx 3.0 upgrade

fabiobaltieri, 2023-05-26:
**Describe the bug**
Hi, looks like `qdec_nrfx` also took a hit with the hal upgrade (https://github.com/zephyrproject-rtos/zephyr/pull/57259), it seems like right now the accumulator is cleared continuously instead of only only on channel get.

I was able to reproduce it on an nrf52840dk with the `samples/sensor/qdec`, board wired for the emulated data (P0.03 P0.04 connected to P0.30 P0.31):

```
$ git checkout origin/main
HEAD is now at 5ab14730f5 sensors: Add async read/decode documentation
$ west update
$ west build -t flash -p -b nrf52840dk_nrf52840 samples/sensor/qdec
```

Shell keeps showing mostly `Position = 0 degrees`

```
$ git checkout 5da98e9d569274aec509ce568978a5cc87417cd2^
$ west update
$ git cherry-pick b96f6fd618905f9f141e99b824fd3f8b9f14e7cd
$ west build -t flash -p -b nrf52840dk_nrf52840 samples/sensor/qdec
```

Shell keeps showing: `Position = 30 degrees`

Verified it on a device with an actual encoder as well. Could you look into it?

**To Reproduce**
See the steps above.

**Expected behavior**
The sensor device should hold the accumulated step count until the get function is called.

**Impact**
It's a regression on QDEC on nRF devices.

**Environment (please complete the following information):**
Linux, Zephyr SDK, 5ab14730f5

fabiobaltieri, 2023-05-26:
CC @awojasinski-nordicsemi

anangl, 2023-05-30:
@fabiobaltieri I was not able to get stable readouts (BTW shouldn't the sample rather output "Rotation" instead of this misleading "Position"?) after applying:
```
$ git checkout 5da98e9d569274aec509ce568978a5cc87417cd2^
$ west update
$ git cherry-pick b96f6fd618905f9f141e99b824fd3f8b9f14e7cd
$ west build -t flash -p -b nrf52840dk_nrf52840 samples/sensor/qdec
```
unless I changed the sampling period in `qdec_nrfx.c` to 512 us or I changed a bit the way the qdec sensor sample performed the emulation so that instead of:
[image]
it was:
[image]

I included a commit with the latter change in #58420.

fabiobaltieri, 2023-05-30:
> @fabiobaltieri I was not able to get stable readouts (BTW shouldn't the sample rather output "Rotation" instead of this misleading "Position"?) after applying:

Interesting, worked for me somehow but nice, thanks for fixing that as well!

PR #58420 (2023-05-30): drivers: sensor: qdec_nrfx: Add missing REPORTRDY interrupt activation
This is a follow-up to commit 09fa46ee4ec9ceea89b45eecaf1b8b984cc25fb8.

Before nrfx 3.0, the QDEC interrupt on REPORTRDY event was activated implicitly when a `reportper` value other than `DISABLED` was used.
Now, the `reportper_inten` field needs to be used to activate this interrupt.

Fixes #58336.

Also correct a bit the encoder emulation performed by the qdec sensor sample.
closed 2023-05-30 by carlescufi