zephyrproject-rtos/zephyr #58359: Segger SystemView can't be used with LwM2M

nixward, 2023-05-27:
**Describe the bug**
Segger SystemView U8, U16, U32 macros conflict with Zephyr LwM2M library macros. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add the following to samples/net/lwm2m_client/prj.conf
```
CONFIG_TRACING=y
CONFIG_SEGGER_SYSTEMVIEW=y
```
3. `west build -b nrf52840dk_nrf52840 samples/net/lwm2m_client -- -DCONF_FILE="prj.conf overlay-ot.conf"`

**Expected behavior**
It should build but doesn't.

**Impact**
Annoyance - requires patch of Zephyr.

**Logs and console output**
```
In file included from /zephyr/subsys/net/lib/lwm2m/lwm2m_obj_device.c:23:
/zephyr/subsys/net/lib/lwm2m/lwm2m_object.h:174:24: error: 'LWM2M_RES_TYPE_unsigned' undeclared here (not in a function); did you mean 'LWM2M_RES_TYPE_NONE'?
  174 |           .data_type = LWM2M_RES_TYPE_ ## _type, }
      |                        ^~~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - MacOS
 - Zephyr SDK
 - Zephyr fc73d6a7b9ddcbbb73898d51d1dfc30a20a674b5

nixward, 2023-09-05:
It's still an issue unfortunately @fabiobaltieri

fabiobaltieri, 2023-09-05:
cc @rlubos anything that could be done there?

rlubos, 2023-09-18:
AFAIK the conclusion was this should be fixed on the Segger side (https://github.com/zephyrproject-rtos/zephyr/pull/58360#issuecomment-1569820299)

closed 2023-12-02 by github-actions[bot]