zephyrproject-rtos/zephyr #58392: Zephyr kernel deadlocks in thread metric benchmark for SMP=2

plmakik, 2023-05-29:
**Describe the bug**

When running thread metric benchmarking tests:
https://github.com/azure-rtos/threadx/tree/master/utility/benchmarks/thread_metric
ported to Zephyr, we observe kernel behavior which looks like an internal synchronization deadlock for two test cases.
Attached are zipped code and screen-shots from debug session.

**To Reproduce**
The failing test cases are selected when in tm_api.h one of two #defines is true 

#define TM_PREEMPT      (1)
or
#define TM_IRQ_PREEMPT  (1)

Our hardware platform is based on Xilinx SOC (ARM Arch64). 
We replicated the problem with kernel version 3.3.99 at Xilinx ZCU102 evaluation board with ATF boot for two cores (SMP=2).

**Expected behavior**
No deadlock when scheduling task

**Impact**
showstopper for our application

**Logs and console output**
Attached are zipped code and screen-shots from debug session.

**Environment (please complete the following information):**
-- west build: building application
[1/124] Generating include/generated/version.h
-- Zephyr version: 3.3.99 - build: e00e895ec275
Windows build 

**Additional context**
We replicated the problem with kernel version 3.3.99 at Xilinx ZCU102 evaluation board with ATF boot for two cores (SMP=2).

[TM_PE_crash.pdf](https://github.com/zephyrproject-rtos/zephyr/files/11592119/TM_PE_crash.pdf)

[zephyr_tm_bug_v1.pdf](https://github.com/zephyrproject-rtos/zephyr/files/11592144/zephyr_tm_bug_v1.pdf)

[tm.zip](https://github.com/zephyrproject-rtos/zephyr/files/11592152/tm.zip)

@abb-bartekf

nashif, 2023-05-29:
@peter-mitsis  can you please take a look?

peter-mitsis, 2023-05-29:
Will do.

peter-mitsis, 2023-06-08:
Taking a look at this today. Apologies for the delay.

peter-mitsis, 2023-06-09:
@plmakik - Would you be able to re-run this test on your end? There have been some recent changes to the scheduling code very close to the time that this bug was filed, and it does not look like your logs reflect those changes. It is possible that they may resolve your issue. If not, then it is back to digging. :)

plmakik, 2023-06-12:
Yes, sure.
I will fetch the latest kernel and re-run TM tests on our platform.
But it may take a few days -  I will post another message with the outcome when ready.

ABB Sp. z o.o. z siedzib? w Warszawie, adres: Warszawa 04-713, ul. ?ega?ska 1, wpisana do Rejestru Przedsi?biorc?w Krajowego Rejestru S?dowego prowadzonego w S?dzie Rejonowym dla m. st. Warszawy, XIII Wydzia? Gospodarczy Krajowego Rejestru S?dowego pod nr KRS 0000004745, nr REGON: 010017168, nr NIP 526-030-44-84, nr BDO: 000011234, kapita? zak?adowy 245 460 765,92 z?.
________________________________
ABB Sp. z o.o. with registered seat at 1 ?eganska Street, 04-713 Warsaw, Poland, registered in the Register of Entrepreneurs of the Polish Court Register maintained by the District Court for the Capital City of Warsaw, XIII Economic Department, under KRS No. 0000004745, REGON No. (statistical number): 010017168, NIP No. (taxpayer identification number) PL5260304484, BDO No. (WEEE registration number) 000011234, share capital: 245 460 765,92 PLN.

plmakik, 2023-06-15:
I have made the test with the latest kernel and the deadlock problem looks pretty much the same.
Please find the report attached.
[zephyr_tm_bug_v2.pdf](https://github.com/zephyrproject-rtos/zephyr/files/11759713/zephyr_tm_bug_v2.pdf)

peter-mitsis, 2023-06-19:
I have been running the test case on qemu_x86_64, and I think that I have found a likely candidate to the cause of the hang. 

The routine k_thread_suspend() currently has a small window where the schedule lock has been released and when the rescheduler is called via z_reschedule_unlocked(). When I plug that hole, I observe the following output ...

```
*** Thread-Metric Preemptive Scheduling Test **** Relative Time: 3
ERROR: Invalid counter value(s). Preemptive counters should not be more that 1 different than the average!
Time Period Total:  2852824

**** Thread-Metric Preemptive Scheduling Test **** Relative Time: 6
ERROR: Invalid counter value(s). Preemptive counters should not be more that 1 different than the average!
Time Period Total:  2834590

**** Thread-Metric Preemptive Scheduling Test **** Relative Time: 9
ERROR: Invalid counter value(s). Preemptive counters should not be more that 1 different than the average!
Time Period Total:  2845557

...
```
Errors are clearly being reported. At this point, I suspect that these errors are related to the test itself, but I will need to be look more closely before I post this potential fix for review.

plmakik, 2023-06-27:
The proposed fix has been tested on our platform and it seems to work fine! We had some hopes for a review of the two-stage locking architecture in the SMP scheduler which always has some deadlock potential, but this perhaps was too much. Thanks a lot!

plmakik, 2023-07-06:
Obviously, in SMP mode there is a race in the code. Therefore it works also as a stress test for us.
But, for a race in application code, one would simply expect an unpredicted behavior of the application rather than kernel deadlock.
We use Thread Metric code to benchmark other RTOSes and only Zephyr ends in this way.
So if the decision is to throw a kernel exception in case the kernel thinks that application tries to do something that the Zephyr kernel considers as unsupported, may be also good for us.
The discussion of why it can't be handled is rather obscured, especially about promises Zephyr gives and others don't. 
Probably nobody expects that k_thread_suspend() preempts the currently running thread on the other core, but there can be a logic that will simply register the intent, which will be effective during the next rescheduling and requires
k_resume() before returning it to the ready queue.  
Anyway, simple graph with thread state transitions and information on what is legal and what is not could help here.

plmakik, 2023-07-14:
A correction about expected behavior.
The synchronous suspend is actually implemented in competing implementations.
If the task runs on the other core, it is preempted by issuing IPI.

peter-mitsis, 2023-08-09:
Update: I am working on a better solution than what was originally posted in #59400 to allow thread suspension on SMP.

peter-mitsis, 2023-10-23:
Missed the stale warning. There is an open PR that should address this.

PR #59400 (2023-11-06): kernel: Change locking in k_thread_suspend()
Updates k_thread_suspend() to work on SMP systems. It functions much like k_thread_abort().

If the target thread is running on another CPU, the target thread gets marked as "halting with a reason". Meanwhile the calling thread waits for the target thread to finish halting (suspend or abort). Note that the reason bit has been implemented such that an abort takes precedence over a suspend to cover the case a thread running on another CPU gets suspended and then aborted (or vice versa). If the target thread is not running on another CPU, then the targ [... cut]
closed 2023-11-06 by nashif