zephyrproject-rtos/zephyr #58394: drivers: dma: emulated dma driver

cfriedt, 2023-05-29:
## Introduction

We propose adding a DMA Emulation driver (Enhancement Issue: #57129, PR: #58375)

As is the case with other driver APIs, a `dma_emul` driver could serve several purposes. Namely,
- providing an ideal / model driver for reference
- potential for configurable backends
- seamless integration with device tree
- multi-instance, etc, for all supported boards
- fast regression testing of app and library code

This RFC should be of interest to collaborators and contributors and maintainers of DMA drivers, Memory Controller drivers, as well as other other drivers that may make use of DMA devices.

### Problem description

The main motivation for adding `dma_emul` is to enable fast functional testing of application and library code, as well as fast functional testing of other emulation devices that make use of Zephyr DMA API.

As it stands currently, it is difficult to reason how to use the DMA API without hardware availability.

Some additional documentation could improve that situation but documentation is also always prone to gaps in human upkeep, etc.

An emulation driver is good in the way that it can formally document the model behaviour in a programmatic way. With that, the interpretation of the doc / model is continuously tested in CI.

Additionally, emulation drivers typically allow for configurable backends - e.g. `gpio_emul` allows users to provide "wiring", the Fuel Gauge backend is now getting configurable too (#58346). 

### Proposed change

- [ ] create `zephyr,dma-emul.yaml` device tree bindings
- [ ] create `drivers/dma/dma_emul.c` device driver
- [ ] add support for `zephyr,dma-emul` to `native_posix` and `native_posix_64`
- [ ] run the set of `tests/drivers/dma/` testsuites in CI regularly, via the `dma_emul` driver
- [ ] add hardware-specific drivers to a `build_all` testsuite
- [ ] improve documentation for the DMA API
- [ ] iron-out any inconsistencies or API topics that are unclear

## Detailed RFC

### Proposed change (Detailed)

Code changes are in the PR below
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/58375

- [ ] Come up with a list of desired DMA Emulator backends
- [ ] Add relevant backend API to PR above

### Dependencies

### Concerns and Unresolved Questions

Some potential things that seemed somewhat unclear or inconsistent
- [ ] `complete_callback_en` seemed to be unused / assumed? in some of the `tests/drivers/dma/` suiites
- [ ] what state should DMA descriptors should be in after starting / stopping / restarting?
  - should restarted transfers start at the beginning or only transfer the remainder?
  - should both options be supported? Is there a way to differentiate?
- [ ] error `status` codes when `error_callback_en` is set

Some additional things that may be good to document in the DMA API
- [ ] A state diagram would be nice to visualize allowable call orders configure, start, stop, reload, etc.
- [ ] It would be good to clarify "memory requirements" for dma controllers based on `dma-channels` and `dma-requests`
  - some offline discussion implied these should form a 2D array of sorts, but it's possible that controllers might also just allocate from an internal pool
- [ ] Are there any known limitations of the DMA API? E.g. is it only intended for "embedded" DMA controllers, or does it scale up to server class?

There is a very strong use case in the server space to use DMA for RPC, that has already happened to a large extent with demos by people such as @kgugala from AntMicro using OpenAMP ([ZDS Talk](https://youtu.be/qiJxki4TPVA), [LPC Talk](https://youtu.be/wC0vaakNg6g)). Particularly salient with the recent addition of the /dev/accel major in Linux. I wonder if there is potentially room for some in-tree samples for that type of application? Even without focusing on OpenAMP, a submission queue + completion queue example could be very useful.

The PR below may be somewhat relevant.
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/48865

## Alternatives

cfriedt, 2023-05-29:
@carlescufi - not sure if I need to add `area: Architecture` to get this into the arch meeting, or if perhaps it should go to a different meeting.

teburd, 2023-05-30:
To clarify/reiterate my stance. I'm fine with this, but I think the emulated driver needs to emulate a specific piece of hardware and how it behaves rather than trying to specify how the DMA API should behave in an ideal/general way. There is no ideal/general way the DMA API should behave.

When emulating for example sensors, we don't expect a emulated generic 6 axis imu driver with device tree properties to describe behavior that applies to all 6 axis sensors. We have a specific sensor being emulated, e.g. bmi160, icm42688, etc. To me, this is how DMA emulation should be done. Emulating specific DMA controllers and their behavior. By the time you try and account for all the variety here you will end up with what amounts to an option per controller anyways, so may as well start that path from the get go.

cfriedt, 2023-06-01:
@teburd

> There is no ideal/general way the DMA API should behave.

Respectfully, I disagree. The entire point of any driver API is to provide an interface to control devices of the same class so that they behave in predictable ways within assumed (and ideally verifiable) boundaries. And actually, in the specific case of DMA, there is a very significant reason for that.

I would be supportive if someone wanted to use the `dma_emul` driver to emulate a particular DMA device as well.

Not all behaviour is practical to specify via device tree either. That is one of the many reasons to support configurable backends.

cfriedt, 2024-01-10:
I think this is finished now

closed 2024-01-10 by cfriedt