zephyrproject-rtos/zephyr #58397: ARC: NSIM: shell is stuck in SMP build for nsim_hs

jacoba-at, 2023-05-29:
**Description**:
When I build the Zephyr shell example application for board nsim_hs_smp, the shell is stuck.
It works fine for board nsim_hs.

**Environment**:
branch: v3.3-branch, hash: ff79476b11
toolchain: MWDT 2022.12

**To reproduce**:
west build -b nsim_hs_smp samples/subsys/shell/shell_module
west flash

**Output**:
all I get (after some mdb output):
ua

**Impact**:
We need to run our software in SMP mode, and we were hoping to rely on NSIM until we real HW to our developers. This hinders our development progress.

evgeniy-paltsev, 2023-05-30:
Just for note - it works without any issue in current main (5ab14730f5) (tested with MWDT 2022.12 and Zephyr SDK 0.16.1):

```
$ west build -b nsim_hs_smp samples/subsys/shell/shell_module
$ ninja -C build -v run
    [***snip***]
uart:~$ version
Zephyr version 3.3.99
uart:~$
uart:~$ help
Please press the <Tab> button to see all available commands.
You can also use the <Tab> button to prompt or auto-complete all commands or its subcommands.
You can try to call commands with <-h> or <--help> parameter for more information.

Shell supports following meta-keys:
  Ctrl + (a key from: abcdefklnpuw)
  Alt  + (a key from: bf)
Please refer to shell documentation for more details.

Available commands:
  bypass              :Bypass shell
  clear               :Clear screen.
  date                :Date commands
    [***snip***]
````

Need to ckeck if there are any issues with older version.

evgeniy-paltsev, 2023-05-30:
The rootcause is the missing 16e9362ba07deedc28de6176ee3efba390b73272 fix in v3.3-branch.

You can add it locally with:

```
git cherry-pick 16e9362ba0
```

With the fix it works fine when built from v3.3-branch:

```
$ ninja -C build -v run
    [***snip***]
uart:~$ version
Zephyr version 3.3.0
uart:~$
```

jacoba-at, 2023-05-31:
Indeed this fixes the issue. 
Thanks!

PR #58456 (2023-07-04): [Backport v3.3-branch] ARC: SMP: fix IDU mask setup
Backport d4d0907ede61fceb1e2200ee2867b9a4371917f1~2..d4d0907ede61fceb1e2200ee2867b9a4371917f1 from #55956.

Fixes: #58397
closed 2023-08-23 by evgeniy-paltsev