zephyrproject-rtos/zephyr #58398: ARC: NSIM: k_sleep function return immediately in nsim_hs board

jacoba-at, 2023-05-29:
**Describe the bug**
When I build samples/hello_world  for nsim_hs board with the the following lines :
printk("Hello World! %s\n", CONFIG_BOARD);
k_sleep(K_MSEC(50000));
printk("After sleep \n");cat cat sam

the function k_sleep(0 does not add any delay;

This can also be observed with building the application samples/subsys/shell/shell_module, and in the shell issue the command:
kernel sleep 50000

which returns immediately.

**To Reproduce**

1. Add the lines above to samples/hello_world
2. west build -b nsim_hs samples/hello_world
3. west flash

**Expected behavior**
Some kind of delay should be observed on the screen. increasing the sleep time does not help.

**Impact**
We resort to busy wait loop instead which hogs the CPU.

**Environment (please complete the following information):**
 - OS: Ubuntu 20
 - Toolchain MWDT 2022.12
 - Branch: v3.3-branch, hash: ff79476b11

evgeniy-paltsev, 2023-05-30:
Luckily / unfortunately this doesn't related to any Zephyr issue but rather how nSIM simulator works it's not cycle accurate (and it never advertised as cycle accurate).

From Zephyr perspective we just setup timer limit and execute `sleep` instruction (I've checked that we don't bump in any corner case here), but in case of nSIM this sleep is just executed *really* fast.

jacoba-at, 2023-05-31:
@evgeniy-paltsev I'm not expecting it to be cycle accurate, but I except to see that the thread goes to sleep, and woken by timer interrupt, which I believe is supported by nsim.
When I applied 16e9362 I see that both in the shell example (cmd: kernel sleep 5000) and hello_world sample, when I build for nsim_hs_smp, indeed there is a delay of about 5 seconds, but when I build for nsim_hs, the call return immediatly, even if I bump the number to 500000.

closed 2023-10-20 by github-actions[bot]