zephyrproject-rtos/zephyr #58402: Bluetooth Host: struct bt_l2cap_server: Missing context argument in accept() callback

donatieng, 2023-05-29:
When registering a L2CAP server with `bt_l2cap_server_register()`, each connection request is accepted/declined in the `int (*accept)(struct bt_conn *conn, struct bt_l2cap_chan **chan)` callback which is a member of the `struct bt_l2cap_server` structure passed upon registration.

This is fine when writing a module which will manage a single L2CAP PSM, however this won't work when multiple servers/PSMs are managed at runtime, as the `accept()` callback is not given any context about which server or PSM it is dealing with.

Workarounds are:
* Use a callbacks table with a fixed number of entries, with a maximum size set at compile time
* Use an architecture dependent functor to add an extra context argument on top of the existing API (this might not work when using an MPU)

Zephyr change options:
* Modify the existing `accept()` callback to pass the PSM as an extra argument
* Modify the existing `accept()` callback to pass the relevant `struct bt_l2cap_server*` as an argument (an associated context can then be recovered using `offsetof()` or similar)

I'd ideally like to contribute a PR for one of the two solutions below, with the last one being my personal preference - but looking for some advice as to which would be preferable!

Cheers

alwa-nordic, 2023-05-31:
I think your preferred solution sounds good!

> Modify the existing accept() callback to pass the relevant struct bt_l2cap_server* as an argument

jhedberg, 2023-06-01:
@donatieng feel free submit a PR with your proposal, it sounds like a good improvement.

jhedberg, 2023-06-01:
And to be clear, the option of adding `struct bt_l2cap_server *` as a parameter is what I'm specifically referring to

donatieng, 2023-07-18:
I've now submitted a PR: https://github.com/zephyrproject-rtos/zephyr/pull/60536

donatieng, 2023-07-18:
@jhedberg @henrikbrixandersen @alwa-nordic would it be possible to add the `RFC` label to this issue please?

henrikbrixandersen, 2023-07-18:
> @jhedberg @henrikbrixandersen @alwa-nordic would it be possible to add the `RFC` and `Stable API Change` labels to this issue please?

Added.

Thalley, 2023-07-21:
I'm not 100% convinced this is a necessary change. 

```c
static int server_accept(struct bt_conn *conn, struct bt_l2cap_server *server, struct bt_l2cap_chan **chan)
{
    ...
}

l2cap_server_a.accept = server_accept;
l2cap_server_b.accept = server_accept;
```

Would be the same as 

```c
static int server_accept(struct bt_conn *conn, struct bt_l2cap_server *server, struct bt_l2cap_chan **chan)
{
    ...
}

static int server_accept_a(struct bt_conn *conn, struct bt_l2cap_chan **chan)
{
    return server_accept(conn, &l2cap_server_a, chan);
}

static int server_accept_b(struct bt_conn *conn, struct bt_l2cap_chan **chan)
{
    return server_accept(conn, &l2cap_server_b, chan);
}

l2cap_server_a.accept = server_accept_a;
l2cap_server_b.accept = server_accept_b;
```

So it does not really solve anything and is just a minor optimization that breaks a stable API that hasn't been changed in at least 8 years.

carlescufi, 2023-08-03:
@donatieng in order to follow the Stable API Change procedure as described [here](https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-incompatible-changes):

- Clean up the PR and make CI pass
- Send an email to the mailing list
- Bring the topic to the Architecture WG meeting (I can help out with this)

In addition to that, please check if there are additional similar callbacks in the Bluetooth API that you think need a similar change for consistency.

donatieng, 2023-08-03:
Thanks for the guidance @carlescufi - I'll clean up the PR in the coming days and review potential other related changes before hitting the mailing list. Thanks for the offer on the help with the WG.

@Thalley you're right that for many use cases this change in un-necessary, however if you create L2CAP servers dynamically, you'll essentially have to use some kind of hack such as maintaining a predetermined-size table of callbacks which would waste memory and make your code less readable.

donatieng, 2023-08-08:
Notes on other callbacks in the Bluetooth host stack:

## Other L2CAP callbacks
* L2CAP channel callbacks all have a relevant `struct bt_l2cap_chan *chan` parameter - as the user is creating these channel structures, user-defined data can be colocated and recovered using the `offsetof()` macro or similar
* L2CAP Fixed channels: All of these are internal to the host stack, and statically defined - therefore I don't think adding a context argument would make sense

## GATT callbacks
* All callbacks have a `void *user_data` or `const struct bt_gatt_attr *attr` parameter (and the `struct bt_gatt_attr` structure contains a user-configurable parameter)

## Connection Management callbacks
This is the main other area I've found which would benefit from letting the user provide some context. All callbacks have a `struct bt_conn *conn` parameter, however the stack is responsible for creating connection objects - and the user cannot associate data with these. Therefore the user would need to perform a lookup to recover any contextual data. In this case I would suggest adding a user-settable `void *user_data` parameter to `struct bt_conn`, however this might be out-of-scope for this RFC. This wouldn't be a breaking change though.

donatieng, 2023-08-08:
## RFCOMM
The RFCOMM callback API seems to have been designed similarly to the L2CAP API. The `accept()` callback from `struct bt_rfcomm_server` doesn't let the user recover any contextual information: `int (*accept)(struct bt_conn *conn, struct bt_rfcomm_dlc **dlc)`. Mirroring the proposed changes would make sense here.

## Isochronous channels
Similar issue. The `struct bt_iso_server`'s accept callback has the following signature: `int (*accept)(const struct bt_iso_accept_info *info, struct bt_iso_chan **chan)`. I don't know much about the audio stack though so not sure if this would be a desirable change here.

Thalley, 2023-08-14:
@donatieng You have a good point w.r.t. dynamic registration. I had not considered that. 

You already mentioned it yourself w.r.t. `user_data` in the callback, which is in fact something we have in our API guidelines: https://docs.zephyrproject.org/latest/develop/api/design_guidelines.html#using-callbacks

So perhaps rather than adding a pointer to the server in the callback, perhaps the register function (or the `struct bt_l2cap_server`) and the callback should get a `user_data` argument. That seems more flexible than just adding a pointer to the server, and adheres to our API guidelines for callbacks. (If the `user_data` is added to the server itself, then it will be equally flexible if the server was returned).

donatieng, 2023-08-22:
I agree that having `user_data` would be even "cleaner" IMO, good point @Thalley.

donatieng, 2023-08-22:
I'll send out an email to the Zephyr mailing list shortly.

donatieng, 2023-08-25:
I just re-read the callback guidelines (thanks for providing the link above, @Thalley). The following paragraphs seem relevant:

> The first parameter should be a pointer to the object most closely associated with the callback. In the case of device drivers this would be const struct device *dev. For library functions it may be a pointer to another object that was referenced when the callback was provided
> An exception to providing user_data as the last parameter may be allowed when the callback itself was provided through a structure that will be embedded in another structure. An example of such a case is [gpio_callback](https://docs.zephyrproject.org/latest/hardware/peripherals/gpio.html#c.gpio_callback), normally defined within a data structure specific to the code that also defines the callback function. In those cases further context can accessed by the callback indirectly by [CONTAINER_OF](https://docs.zephyrproject.org/latest/kernel/util/index.html#c.CONTAINER_OF).

I also re-reviewed the wider L2CAP API. One a L2CAP channel is opened, callbacks with similar signatures signal various events - e.g. `void (*connected)(struct bt_l2cap_chan *chan)`. In this case, wider context can be recovered using `CONTAINER_OF` or `offsetof` as the user is managing the `struct bt_l2cap_chan` structure.

donatieng, 2023-08-25:
Added the `void *user_data` alternative option in the issue's description above

donatieng, 2023-08-25:
@carlescufi If you're happy with the state of this proposal do you think we could bring it to an upcoming WG meeting?

carlescufi, 2023-09-06:
> @carlescufi If you're happy with the state of this proposal do you think we could bring it to an upcoming WG meeting?

@donatieng can you join the next meeting tomorrow? https://lists.zephyrproject.org/g/bluetooth/calendar

donatieng, 2023-09-06:
Sounds great, happy to!

closed 2023-09-14 by carlescufi