zephyrproject-rtos/zephyr #58405: doc: pictures in pdf documentation showing "[Not supported by viewer]"

botbotty, 2023-05-30:
**Describe the bug**
I downloaded a copy of [zephyr documentation in pdf format](https://docs.zephyrproject.org/latest/zephyr.pdf)

Some of the pcitures in the pdf documentation are not showing correct, for example:

PDF:
[image]

same picture in https://docs.zephyrproject.org/latest/kernel/drivers/index.html:
[image]

Searching for "Not supported by viewer" in the pdf file gives one more such picture:
[image]

I have tried open the pdf in both adobe acrobat reader and microsoft edge, results are the same.

kartben, 2023-06-01:
Looks like rsvg converter has limitations. I was initially thinking I was going to fix the two images you reported as problematic, as an easy fix, but it would appear there are many others that are messed up. See e.g. 

[image]

which renders as

[image]

I am going to see if switching to a different renderer could help (https://pypi.org/project/sphinxcontrib-svg2pdfconverter/).

kartben, 2023-06-01:
This is an issue with diagrams made using draw.io/diagrams.net, see explanations here: https://www.drawio.com/doc/faq/svg-export-text-problems. 
There doesn't seem to be an easy fix besides reworking all the diagrams ; either have them export text labels as SVG shapes, or disable formatted text and word wrapping. I'd prefer option 2 since option 1 can make files with a lot of text *much* bigger. Long term solution would be to provide better guidelines on do's and don'ts for SVG image, and have something like mermaid for simpler diagrams that can be described as text.

FWIW here's a list of the files causing issues:

```bash
$ grep "requiredFeatures=\"http://www.w3.org/TR/SVG11/feature#Extensibility" . -Rl | sort
./doc/build/cmake/build-build-phase-1.svg
./doc/build/cmake/build-build-phase-2.svg
./doc/build/cmake/build-build-phase-3.svg
./doc/build/cmake/build-build-phase-4.svg
./doc/build/cmake/build-build-phase-5.svg
./doc/build/cmake/build-build-phase-6.svg
./doc/build/cmake/build-config-phase.svg
./doc/build/cmake/build-postprocess-1.svg
./doc/build/cmake/build-postprocess-2.svg
./doc/build/cmake/build-postprocess-3.svg
./doc/build/cmake/build-postprocess-4.svg
./doc/build/dts/zephyr_dt_build_flow.svg
./doc/build/dts/zephyr_dt_i2c_example.svg
./doc/build/dts/zephyr_dt_i2c_high_level.svg
./doc/build/dts/zephyr_dt_inputs_outputs.svg
./doc/build/sysbuild/sysbuild.svg
./doc/connectivity/bluetooth/api/mesh/images/blob_srv.svg
./doc/connectivity/bluetooth/api/mesh/images/dfu_srv.svg
./doc/connectivity/bluetooth/img/ble_audio_arch.svg
./doc/connectivity/bluetooth/img/cap_proc.svg
./doc/connectivity/bluetooth/img/gaf.svg
./doc/connectivity/bluetooth/img/zephyr_gaf.svg
./doc/contribute/media/ext-src-flowchart.svg
./doc/develop/test/fixtures.svg
./doc/develop/test/twister_and_pytest.svg
./doc/hardware/peripherals/1-Wire_bus_topology.drawio.svg
./doc/kernel/drivers/device_driver_model.svg
./doc/kernel/services/scheduling/cooperative.svg
./doc/kernel/services/scheduling/preemptive.svg
./doc/kernel/services/scheduling/timeslicing.svg
./doc/kernel/services/threads/priorities.svg
./doc/kernel/services/threads/thread_states.svg
./doc/project/lts.svg
./doc/project/release_cycle.svg
./doc/services/pm/images/central_method.svg
./doc/services/pm/images/system-pm.svg
./doc/services/portability/posix.svg
./samples/bluetooth/mtu_update/img/mtu.svg
```

closed 2023-10-25 by github-actions[bot]