zephyrproject-rtos/zephyr #58416: tests/kernel/sleep failed with cortex M0+ stm32 devices

FRASTM, 2023-05-30:
**Describe the bug**
when running the tests/kernel/sleep usecase on the stm32 M0+ mcu, the testcase fails on sleep.test_usleep  with kernel oops : 
```
START - test_usleep                                                             
START - test_usleep                                                             
E: Bad system call id 1072 invoked                                              
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000                   
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000                   
E:  xpsr:  0x00000000                                                           
E: Faulting instruction address (r15/pc): 0x9c8e3d41                            
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0                               
E: Current thread: 0x20000270 (test_usleep)                                     
E: Halting system       
```

Please also mention any information which could help others to understand
the problem you're facing:
 - nucleo_g071rb or nucleo_l073rz boards which are cortex M0+  
 the pb is not seen with cortex M0 or other cortex 

 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
-->   This is very annoying to revert commit  bdf2e56ce1a1fd0e60e3595d49aac958cc5d100f  and see the test PASSED
because this has nothing to do with the pb.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b nucleo_l073rz tests/kernel/sleep/ or west build -p always -b nucleo_g071rb tests/kernel/sleep/
2. west flash
3. See error

**Expected behavior**
```
START - test_usleep                                                             
elapsed_ms = 1000                                                               
 PASS - test_usleep in 1.003 seconds                                            
                                                                                
```

**Impact**
CI failure 

**Logs and console output**
```
*** Booting Zephyr OS build v3.4.0-rc1 ***                                      
Running TESTSUITE sleep                                                         
===================================================================             
START - test_sleep                                                              
Kernel objects initialized                                                      
Test thread started: id = 0x200001b8                                            
Helper thread started: id = 0x20000100                                          
Testing normal expiration of k_sleep()                                          
Testing: test thread sleep + helper thread wakeup test                          
Testing: test thread sleep + isr offload wakeup test                            
Testing: test thread sleep + main wakeup test thread                            
Testing kernel k_sleep()                                                        
 PASS - test_sleep in 2.032 seconds                                             
===================================================================             
START - test_sleep_forever                                                      
Kernel objects initialized                                                      
 PASS - test_sleep_forever in 0.003 seconds                                     
===================================================================             
START - test_usleep                                                             
E: Bad system call id 1072 invoked                                              
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000                   
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000                   
E:  xpsr:  0x00000000                                                           
E: Faulting instruction address (r15/pc): 0x9c8e3d41                            
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0                               
E: Current thread: 0x20000270 (test_usleep)                                     
E: Halting system                                                               

```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain : Zephyr SDK 0.16.1
 - Commit v3.4.0-rc1  SHA 183e0f854b83de58e7565b297d2af95e52e6a54c

**Additional context**

erwango, 2023-06-06:
I doubt this is STM32 specific but rather linked to M0+ arch.

@nandojve @dleach02  Can you see something similar on NXP/Atmel platforms ?

nandojve, 2023-06-06:
Hi @erwango ,

I tested using this hash:
135f5376c1 - ci: workflows: create cache directory for ccache

```
west build -p always -b atsamd20_xpro tests/kernel/sleep

*** Booting Zephyr OS build zephyr-v3.0.0-1344-g55ede2fe1202  ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x200000c0
Helper thread started: id = 0x20000030
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
 PASS - test_sleep in 2.032 seconds
===================================================================
START - test_sleep_forever                                                      
Kernel objects initialized                                                      
 PASS - test_sleep_forever in 0.003 seconds                                     
===================================================================             
START - test_usleep                                                             
elapsed_ms = 1500                                                               
 PASS - test_usleep in 1.502 seconds                                            
===================================================================             
TESTSUITE sleep succeeded                                                       
                                                                                
------ TESTSUITE SUMMARY START ------                                           
                                                                                
SUITE PASS - 100.00% [sleep]: pass = 3, fail = 0, skip = 0, total = 3 duration s
 - PASS - [sleep.test_sleep] duration = 2.032 seconds                           
 - PASS - [sleep.test_sleep_forever] duration = 0.003 seconds                   
 - PASS - [sleep.test_usleep] duration = 1.502 seconds                          
                                                                                
------ TESTSUITE SUMMARY END ------
```

```
west build -p always -b atsamr21_xpro tests/kernel/sleep

*** Booting Zephyr OS build v3.4.0-rc2-96-g135f5376c1e6 ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
Kernel objects initialized
Test thread started: id = 0x200000c0
Helper thread started: id = 0x20000030
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
 PASS - test_sleep in 2.032 seconds
===================================================================
START - test_sleep_forever
Kernel objects initialized
 PASS - test_sleep_forever in 0.003 seconds
===================================================================
START - test_usleep
elapsed_ms = 1500
 PASS - test_usleep in 1.502 seconds
===================================================================
TESTSUITE sleep succeeded
                                                                                
------ TESTSUITE SUMMARY START ------                                           
                                                                                
SUITE PASS - 100.00% [sleep]: pass = 3, fail = 0, skip = 0, total = 3 duration s
 - PASS - [sleep.test_sleep] duration = 2.032 seconds                           
 - PASS - [sleep.test_sleep_forever] duration = 0.003 seconds                   
 - PASS - [sleep.test_usleep] duration = 1.502 seconds                          
                                                                                
------ TESTSUITE SUMMARY END ------                                             
                                                                                
===================================================================             
PROJECT EXECUTION SUCCESSFUL
```

erwango, 2023-06-07:
Thanks @nandojve for having done the test.
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/58351 which confirms observations in https://github.com/zephyrproject-rtos/zephyr/issues/58416#issuecomment-1579023037 and also likely confirms `Maybe these harmless changes unveil a weakness in the global M0+ code.`

cfriedt, 2023-06-07:
> After analysis, it appears that ["drivers: Add auxdisplay (text) interface"](https://github.com/zephyrproject-rtos/zephyr/commit/bdf2e56ce1a1fd0e60e3595d49aac958cc5d100f) (which gets the test passed when reverted) actually has an impact on the generated binary. One example:
> 
> ```
> static inline int64_t k_uptime_get(void)
> {
> [...]
> 	return z_arm_thread_is_in_user_mode();
>  80009e6:	f00a fec7 	bl	800b778 <z_arm_thread_is_in_user_mode>
> 	if (z_syscall_trap()) {
>  80009ea:	2800      	cmp	r0, #0
>  80009ec:	d04b      	beq.n	8000a86 <_sleep_test_usleep_wrapper+0xce>
> 	register uint32_t ret __asm__("r0") = arg1;
>  80009ee:	2001      	movs	r0, #1
> +	register uint32_t r6 __asm__("r6") = call_id;
> + 80009f0:	26f9      	movs	r6, #249	; 0xf9
> 	__asm__ volatile("svc %[svid]\n"
> [...]
> ```
> 
> These changes may be harmless, but the fact is that there ia a finally impact on test result. Maybe these harmless changes unveil a weakness in the global M0+ code.
> 
> Note also that there are multiple issues in drivers/auxdisplay/auxdisplay_handlers.c:
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/5d87abccb278cbda114773a5affe703b6bc839d6/drivers/auxdisplay/auxdisplay_handlers.c#L112-L118
> 
> 
> => Mix of backlight_get/brightness_set in the syscall handler definition.
> There are multiple similar issues, but even when fixed, test is still failed. I had a further try, removing any of the function defined in `struct auxdisplay_driver_api` in `include/zephyr/drivers/auxdisplay.h` get the test passed.

@nordicjm - have you seen this?

FRASTM, 2023-08-07:
not reproduced (4d304a2b9eabfb2beba1fc206d1fbc663de446d5)

closed 2023-08-07 by FRASTM