zephyrproject-rtos/zephyr #58424: can anyone help me to sort out this issue..!

hariprasath1163, 2023-05-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

hariprasath1163, 2023-05-30:
[image]

I am using a 2.8 inch TFT display.
while including the #include <display> shows error.
Will you ppl please help to sort it out.

Thankyou.

carlescufi, 2023-05-30:
Please use Discord or any of the other means of help we have. GitHub issues are for bugs or enhancements.

closed 2023-05-30 by carlescufi