zephyrproject-rtos/zephyr #58428: Detaching and attaching SPI and GPIO in driver

nguyenmthien, 2023-05-30:
**Is your feature request related to a problem? Please describe.**
Some peripheral devices require switching pins between SPI and GPIO control during operation. One example is the [TLC5916 ](https://www.ti.com/lit/ds/symlink/tlc5916.pdf) SPI LED driver. During operation it is necessary to change between the special and normal working mode by sending the subsequently mentioned signal sequence:
[image]

This can easily achieved by detaching the CLK pins from the SPI controller, enable GPIO, generate the sequence using GPIO control, disable GPIO, then attaching the pin to the SPI controller again for normal SPI operation. 
Such approach can easily be done in the application by declaring a seperate SPI device with a seperate configuration. However, in the driver, where everything have to be inherited from the devicetree and initialized post kernel, this method is poorly documented and/or unavailable.

**Describe the solution you'd like**
Let driver modules to initialize and de-initialize its submodules during operation. 

**Describe alternatives you've considered**
One workaround is to use bitbanging SPI instead of normal SPI. This approach is less desirable, especially on less powerful SoCs. 

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

henrikbrixandersen, 2023-05-30:
A driver _can_ switch between pin functionality using pinctrl at runtime. See e.g. https://github.com/zephyrproject-rtos/zephyr/blob/86fcc5de9745011dfdd8102a8d6c301652705e12/drivers/i2c/i2c_mcux_lpi2c.c#L243

henrikbrixandersen, 2023-06-06:
> A driver _can_ switch between pin functionality using pinctrl at runtime. See e.g.
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/86fcc5de9745011dfdd8102a8d6c301652705e12/drivers/i2c/i2c_mcux_lpi2c.c#L243

@nguyenmthien Does this solve your issue?

nguyenmthien, 2023-06-07:
> A driver _can_ switch between pin functionality using pinctrl at runtime. See e.g.
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/86fcc5de9745011dfdd8102a8d6c301652705e12/drivers/i2c/i2c_mcux_lpi2c.c#L243

Hi @henrikbrixandersen, thank you for your reply. 
Yes it is correct that your example did show a usecase where pinctrl is used to change pin functionalities upon requests from the driver on runtime. This i2c driver utilities its own pinctrl and to my knowledge, is abstracted away when other peripherals or application use its API. Furthermore, [pinctrl is not intended to switch between peripheral functions (CAN/SPI/I2C) and GPIO](https://docs.zephyrproject.org/latest/hardware/pinctrl/index.html#pin-control-vs-gpio). For the shake of portability of the driver for the aforementioned LED chip, it should use both the abstracted Zephyr-generic GPIO- and SPI - driver.

henrikbrixandersen, 2023-06-12:
> Furthermore, [pinctrl is not intended to switch between peripheral functions (CAN/SPI/I2C) and GPIO](https://docs.zephyrproject.org/latest/hardware/pinctrl/index.html#pin-control-vs-gpio).

You are misreading that part of the documentation. Pinctrl is perfectly fine to use for switching pins between peripheral functions. The GPIO controller drivers, however, switch pins (back) to GPIO function when configuring a pin.

(cc: @gmarull)

nguyenmthien, 2023-06-29:
> You are misreading that part of the documentation. Pinctrl is perfectly fine to use for switching pins between peripheral functions. The GPIO controller drivers, however, switch pins (back) to GPIO [...]

Can you point to an example of such function? I am failing to see how it is implemented.

henrikbrixandersen, 2023-06-29:
> Can you point to an example of such function? I am failing to see how it is implemented.

Yes, I already provided a link to such a function above.

nguyenmthien, 2023-09-01:
@henrikbrixandersen is this a duplicate of #40385?

nguyenmthien, 2023-09-13:
Closed due to duplicate of #40385

closed 2023-09-13 by nguyenmthien