zephyrproject-rtos/zephyr #58435: sockets: offloaded: modem not connecting

WilliamGFish, 2023-05-30:
**ESP-AT modem fails to send data**
After downloading the latest updates the ESP-AT modem no longer send data over IP4 network.

The device connects to the network and appears to have address assigned but is unable to connect to other services; eg SNTP or MQTT.

 - ESP32-AT Modem
 - Have attempted to find change that is effecting connection, have been rolling but not found  specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Showstopper

**Logs and console output**
The target stopped in MemManage exception state.
Reason: Unknown at 0x00000000.
For more information about analyzing faults on Cortex-M, refer to:
https://wiki.segger.com/Cortex-M_Fault.

**Environment (please complete the following information):**
 - OS:  Windows
 - CMake version: 3.25.0-rc2
 - west version "1.0.0"
 - Zephyr version: 3.4.0-rc1 (build: v3.4.0-rc1-50-g147d4067e6de)

rftafas, 2023-05-30:
Question: why is this listed under 'Platform ESP32' if most likely it will be used with other chips than ESP32?

ESP-AT is actually an ESP-IDF application.

WilliamGFish, 2023-05-30:
> Question: why is this listed under 'Platform ESP32' if most likely it will be used with other chips than ESP32?
> 
> ESP-AT is actually an ESP-IDF application.

You're right nothing to do with ESP32 specifically it's a general networking issue. I'm frustrated with the lack of hardware testing.

rlubos, 2023-05-31:
> It should in fact be before the recv not after:
SET_ERRNO(net_context_connect(ctx, addr, addrlen, cb, timeout,
ctx->user_data));
SET_ERRNO(net_context_recv(ctx, zsock_received_cb, K_NO_WAIT,
ctx->user_data));

No, it should not. This was a conscious change, to prevent race condition between connect/recv - see https://github.com/zephyrproject-rtos/zephyr/issues/58084 for the details about the problem.

Native IP stack allows to register recv callback before connection - offloaded implementations should comply with this behavior.

RomainPelletant, 2023-08-04:
Seems to be solved in #59857

carlescufi, 2023-10-04:
Fixed.

closed 2023-10-04 by carlescufi