zephyrproject-rtos/zephyr #58455: Bluetooth: mesh:  Warnings and Fatal Errors when Publishing Large Data from Multiple Nodes

OmarM17, 2023-05-30:
I am currently working on a Bluetooth mesh project using Zephyr RTOS and nRF52840. I have four nodes in the network, where three nodes are publishing payloads periodically to the fourth node's unicast address. Each payload size is 41 bytes, and each of the three nodes sends 8 payloads per second. However, I am encountering several issues in my network, and I need assistance in resolving them.

1. Warnings/Errors in Receiver Node:
`bt_mesh_transport: Ignoring old SeqAuth 0x0...` warning occurs frequently.
`bt_mesh_access: Publication sending took longer than the period` error occurs occasionally.

2. Warnings/Errors in Sender Nodes:
`bt_mesh_transport: No matching TX context for ack` warning occurs frequently.
`bt_mesh_transport: Ran out of retransmit attempts` error occurs occasionally.

3. Fatal Error in Receiver Node:
After approximately 30 min of operation, the receiver node encounters a fatal error. The fatal error log messages vary, but some examples include:
```
 <err> bt_mesh_net: Failed to stor Seq value
 <err> os: ***** USAGE FAULT *****
 <err> os:   Illegal use of the EPSR
---

 <wrn> bt_mesh_transport: Duplicate SDU from src 0x003d
 <err> bt_mesh_rpl: Failed to store RPL bt/mesh/RPL/3d value
 <err> os: ***** USAGE FAULT *****
 <err> os:   Illegal use of the EPSR

----

 <wrn> bt_mesh_transport: Duplicate SDU from src 0x003a
 <err> os: ***** BUS FAULT *****
 <err> os:   Imprecise data bus error

---

 <wrn> bt_mesh_transport: Ignoring old SeqAuth 0x005b6d
 <wrn> bt_mesh_transport: Ignoring old SeqAuth 0x005d18
 <err> os: ***** BUS FAULT *****
 <err> os:   Imprecise data bus error

```

* I made the following modifications in the config file for the receiver node:

```
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_BT_RX_STACK_SIZE=4096
CONFIG_BT_MESH_RPL_STORE_TIMEOUT=600
CONFIG_BT_MESH_CRPL=2000
CONFIG_BT_MESH_MSG_CACHE_SIZE=1000
CONFIG_BT_MESH_ADV_BUF_COUNT=256
CONFIG_BT_MESH_SEG_BUFS=256
CONFIG_BT_MESH_RX_SEG_MAX=10
CONFIG_BT_MESH_TX_SEG_MAX=10
CONFIG_BT_MESH_RX_SEG_MSG_COUNT=100
CONFIG_BT_MESH_TX_SEG_MSG_COUNT=100
CONFIG_BT_MESH_LOOPBACK_BUFS=32
```

*I am seeking guidance on resolving the following issues:*

1. Understanding the cause and resolution for the warnings and errors mentioned above.
2. Determining the reason behind the fatal errors and finding a solution to prevent them.
3. Ensuring the reliable reception of payloads from the sender nodes.

Any assistance, insights, or recommendations regarding these issues would be greatly appreciated.

PavelVPV, 2023-06-01:
First, regarding warnings:

1.a. `bt_mesh_transport: Ignoring old SeqAuth 0x0...` means that the segmented message has been already received by the node some time ago. I think this particular cause may happen when the receiver received a pair of segmented messages in a row, but then receives retransmissions of some segments of the first segmented message from this pair.

1.b. `bt_mesh_access: Publication sending took longer than the period` means that publication period is shorter or close to the time needed to send the message. Since in your case it is segmented message, it may take much longer to complete the previous transmission. Especially if all segments were not confirmed at the first attempt. In this case, the next publication will just start immediately after the previous one completed. If you don't want to see this warning I would first recommend to increase the period time. You may also shorten interval between retransmission of segmented messages and number of retransmissions attempts, but this will reduce probability of reception of the segmented messages.

2.a `bt_mesh_transport: No matching TX context for ack` means that the sender received ack for already completed segmented message transmission. In this case, the receiver sent one ack that confirmed message transmission, but the sender managed to re-send not-confirmed segments before it received this ack (most likely that the ack was lost, less likely that it happened due to wrong timings configuration of segmented message). Then the sender received the ack and completed the transmission. And after that the receiver received the already received segments and re-sent the ack again. The sender received the ack and printed this log message.

2.b. `bt_mesh_transport: Ran out of retransmit attempts` means that transport layer didn't receive ack for the segmented message and aborted transmission.

Regarding this:
> Each payload size is 41 bytes, and each of the three nodes sends 8 payloads per second.

How each of these 8 payload is sent within 1 second if the lowest resolution of Publish Period state is 100ms? Also, since each payload is 41 bytes, the message will consists of 4 segments (12 bytes of actual payload per segment). Transmission of a single segment depends on `CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT` and `CONFIG_BT_MESH_NETWORK_TRANSMIT_INTERVAL` values. So, in average, the transmission of N segments once will take in worst case `(10ms + (CONFIG_BT_MESH_NETWORK_TRANSMIT_INTERVAL + 10) * CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT) * N` (but not less than 20ms between segments). For default configuration, where `CONFIG_BT_MESH_NETWORK_TRANSMIT_INTERVAL` is 20 and `CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT` is 2, for 4 segments message, transmitting all segments will take around 280ms. And this is without retransmission of missing segments. You may decrease it to around 80ms by setting `CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT` to 0 but this will reduce probability of receiving messages if collisions happen and still doesn't consider retransmissions. So my suggestion is to reduce interval between each publication.

Regarding the faults. It would be good if you can get the whole assertion message so that we can potentially see the address that causes the issue. You can also enable `CONFIG_THREAD_NAME` so that thread name is printed in the assertion log. Also, do you have `CONFIG_RESET_ON_FATAL_ERROR` option enabled? If so, please disable it. I can also suggest to enable `CONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU` just in case this issue is related to null pointer dereference.

From what I see in the logs, you have a lot of errors related to storing data to flash and then the assertion happens. Do you store anything else in settings?

I also see you set `CONFIG_BT_MESH_CRPL=2000`. You don't really need that big RPL since you have only 4 nodes in the network. Otherwise you need to increase settings partition size because every RPL entry takes ~35 bytes. For 2000 entries you need ~70kBytes of flash, while default settings partition size is 32kBytes:
https://github.com/zephyrproject-rtos/zephyr/blob/577a7f2fac44cc30740048e0d83b4420635e16e5/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts#L282-L285

Also, since you have only 4 devices in the network, you don't need that many seg msg contexts (`CONFIG_BT_MESH_RX_SEG_MSG_COUNT` and `CONFIG_BT_MESH_TX_SEG_MSG_COUNT`) as only one segmented message can be sent at a time to the same destination. Bigger `CONFIG_BT_MESH_TX_SEG_MSG_COUNT` value you may need to queue outgoing messages. But `CONFIG_BT_MESH_RX_SEG_MSG_COUNT` I think could be reduced to the number of src<->dst pairs expected to be received by a node at a time.

OmarM17, 2023-06-14:
Thank you al lot ! @PavelVPV 
I made some modifications based on your suggestions, which resulted in some improvements. As a result, the frequency of the FATAL ERROR has reduced. However, I'm still struggling to fully understand the underlying problem.

**Sender node changes :**
```
CONFIG_BT_MESH_ADV_BUF_COUNT=255
CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT=1
CONFIG_BT_MESH_NETWORK_TRANSMIT_INTERVAL=10
```
I have also made a change to the periodic publication period, setting it to 200 ms. This means that the 41 bytes of data are now being sent 5 times per second with the NETWORK_TRANSMIT_COUNT=1 and NETWORK_TRANSMIT_INTERVAL=10.

**Receiver node changes:**
```
CONFIG_BT_MESH_SEQ_STORE_RATE=1000000
CONFIG_BT_MESH_RPL_STORE_TIMEOUT=-1
CONFIG_BT_MESH_CRPL=500
CONFIG_BT_MESH_MSG_CACHE_SIZE=1000
CONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU=y
```
I have changed SEQSTORE_RATE and RPLSTORE_TIMEOUT in order to avoid problems caused by the RPL, and I also wonder if I can disable all the RPL features since I am not in need of them.

Here are some example of the fatal errors
```
[00:06:20.534,271] <dbg> bt_mesh_transport: trans_seg: ASZMIC 1 AKF 1 AID 0x04
[00:06:20.534,301] <dbg> bt_mesh_transport: trans_seg: SeqZero 0x09e7 SegO 0 SegN 4
[00:06:20.534,332] <dbg> bt_mesh_transport: seg_rx_alloc: New RX context. Block Complete 0x0000001f
[00:06:20.534,362] <dbg> bt_mesh_transport: trans_seg: Received 0/4
-[00:06:20.582,244] <dbg> bt_mesh_transport: bt_mesh_trans_send: net_idx 0x0000 app_idx 0x0000 dst 0xc002
[00:06:20.582,305] <err> os: ***** BUS FAULT *****
[00:06:20.582,336] <err> os:   Precise data bus error
[00:06:20.582,336] <err> os:   BFAR Address: 0x8020000
[00:06:20.582,366] <err> os: r0/a1:  0x00000000  r1/a2:  0x000357dc  r2/a3:  0x00000002
[00:06:20.582,366] <err> os: r3/a4:  0x08020000 r12/ip:  0x0002d08c r14/lr:  0x00003bd7
[00:06:20.582,397] <err> os:  xpsr:  0x61000000
[00:06:20.582,397] <err> os: Faulting instruction address (r15/pc): 0x00027480
[00:06:20.582,427] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:06:20.582,458] <err> os: Current thread: 0x20009a78 (sysworkq)
[00:06:20.625,244] <err> os: Halting system

---------------------------------------------------------------------------------------

[00:55:30.351,684] <dbg> bt_mesh_net: bt_mesh_net_send: src 0x007e dst 0x007c len 7 headroom 9 tailroom 13
[00:55:30.351,715] <dbg> bt_mesh_net: bt_mesh_net_send: Payload len 7: 006ce00000001f
[00:55:30.351,745] <dbg> bt_mesh_net: bt_mesh_net_send: Seq 0x0f5cbe
[00:55:30.351,776] <dbg> bt_mesh_net: net_header_encode: src 0x007e dst 0x007c ctl 1 seq 0x0f5cbe
[00:55:30.351,898] <dbg> bt_mesh_transport: sdu_recv: AKF 1 AID 0x04
[00:55:30.352,081] <dbg> bt_mesh_transport: sdu_recv: Decrypted (AppIdx: 0x000)
[00:55:30.352,081] <dbg> bt_mesh_access: bt_mesh_model_recv: app_idx 0x0000 src 0x007c dst 0x007e
[00:55:30.352,142] <err> os: ***** BUS FAULT *****
[00:55:30.352,172] <err> os:   Precise data bus error
[00:55:30.352,172] <err> os:   BFAR Address: 0x1000005
[00:55:30.352,203] <err> os: r0/a1:  0x00000000  r1/a2:  0x0003a7c4  r2/a3:  0x00000002
[00:55:30.352,233] <err> os: r3/a4:  0x01000005 r12/ip:  0x00031060 r14/lr:  0x00003f07
[00:55:30.352,233] <err> os:  xpsr:  0x61000000
[00:55:30.352,233] <err> os: Faulting instruction address (r15/pc): 0x0002b43c
[00:55:30.352,264] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:55:30.352,294] <err> os: Current thread: 0x20009760 (BT RX)
[00:55:30.434,448] <err> os: Halting system

------------------------------------------------------------------------------------

[00:34:06.224,792] <wrn> bt_mesh_net: IV Update before minimum duration
[00:34:06.664,581] <err> os: ***** BUS FAULT *****
[00:34:06.664,581] <err> os:   Imprecise data bus error
[00:34:06.664,611] <err> os: r0/a1:  0x2002b058  r1/a2:  0xf02a0bcb  r2/a3:  0x61000000
[00:34:06.664,611] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0002f11b
[00:34:06.664,642] <err> os:  xpsr:  0x6100021b
[00:34:06.664,642] <err> os: Faulting instruction address (r15/pc): 0x0002695a
[00:34:06.664,672] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:34:06.664,672] <err> os: Fault during interrupt handling
[00:34:06.664,703] <err> os: Current thread: 0x20001ac0 (BT TX)
[00:34:06.718,231] <err> os: Halting system
```

And here are the errors and warnings that still occur: 
Receiver node : 
```
<err> bt_mesh_transport: Out of network buffers
<wrn> bt_mesh_transport: Ignoring old SeqAuth 0x0047dd
```
Sender nodes:
```
<err> bt_mesh_transport: Ran out of retransmit attempts
<wrn> bt_mesh_access: Publication sending took longer than the period
<wrn> bt_mesh_transport: No matching TX context for ack
```

Thank you

PavelVPV, 2023-06-16:
> >> addr2line -e build/zephyr/zephyr.elf -a 0x0002b43c
> 0x0002b43c
> C:/zephyrproject/3.1.99/zephyrproject/zephyr/lib/libc/minimal/source/string/string.c:111

Is there a place anywhere where you could potentially pass NULL pointer to `strlen`? https://github.com/zephyrproject-rtos/zephyr/blob/68dc53b0779a37342f86a29362c8efdafdcea005/lib/libc/minimal/source/string/string.c#L111

The other two faults don't really point to any specific line.

Are you using nRF52840 Development Kit? After the fault you can try to run `west debug` to attach debugger to the board, then print backtrace (`bt` in gdb) and check what was causing the fault. You may also try to enable `CONFIG_DEBUG_THREAD_INFO` and see all threads (`info thread` in gdb) after attaching the debugger. Then use `thread <N>` to select specific thread and `bt` to examine backtrace in case the fault is processed in a separate thread.

OmarM17, 2023-06-21:
Thank you for your assistance, I really appreciate it.

I've been trying to troubleshoot some bugs, but I feel like I might have gotten off track. My main issue is to send periodic messages of 41 bytes from multiple sender nodes to a single receiver node, with the shortest period allowed by the Bluetooth mesh.

To simplify things, I've cleaned up both the sender and receiver codes and only kept the definition of a single model in both sides. I've started the periodic publication with a period of 100ms, and I've set some specific mesh parameters in the sender:

```
CONFIG_BT_MESH_TX_SEG_MAX=4

CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT=0
CONFIG_BT_MESH_NETWORK_TRANSMIT_INTERVAL=10

CONFIG_BT_MESH_TX_SEG_RETRANS_COUNT=1
CONFIG_BT_MESH_TX_SEG_RETRANS_TIMEOUT_UNICAST=200

CONFIG_BT_MESH_RELAY=n
CONFIG_BT_MESH_FRIEND=n
CONFIG_BT_MESH_LOW_POWER=n
CONFIG_BT_MESH_MODEL_EXTENSIONS=n
```

For the receiver, I've set all parameters to their default values, except for:

```
CONFIG_BT_MESH_RX_SEG_MAX=4
CONFIG_BT_MESH_FRIEND=n
CONFIG_BT_MESH_LOW_POWER=n
CONFIG_BT_MESH_MODEL_EXTENSIONS=n
CONFIG_BT_MESH_RELAY=n
```

During my testing, I noticed the following:

1. When the sender publishes to the receiver's unicast address with a period of 100ms, the actual publication period is around 400ms. According to the formula (10ms + (CONFIG_BT_MESH_NETWORK_TRANSMIT_INTERVAL + 10) * CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT) * N, it should be only 90ms.

2. I repeated the test, but this time publishing to a group address to avoid the acknowledgement process. According to the mesh profile (page 60), publishing to a group address should not send Segment Acknowledgment messages. However, I found that the actual publication period was around 1.8 seconds.

My questions are:

- How can we understand these behaviors and what parameters should we change to achieve the minimum publication period and the maximum throughput allowed by Bluetooth mesh?
- How can we determine the limits of Bluetooth mesh in such cases?
- Would it be advisable for me to upgrade from Zephyr version 3.0.99 to the latest or a newer version in order to address this problem more effectively?

Any insights or guidance would be greatly appreciated.

Thank you once again for your help.

PavelVPV, 2023-06-22:
> 	Would it be advisable for me to upgrade from Zephyr version 3.1.99 to the latest or a newer version in order to address this problem more effectively?

I will answer on this question first. Yes, you should definitely use the latest zephyr version. There has been many changes since 3.2.0. For example, in Zephyr 3.2.0 mesh still uses the legacy advertising API which is much slower (https://github.com/zephyrproject-rtos/zephyr/blob/80a986f43890273bbd0bf4e2da903089b2d821b9/subsys/bluetooth/mesh/adv_legacy.c#L60-L68) than the extended advertising API. With legacy advertiser even if CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT is set to 0 the lowest advertising duration is 50ms, while with the extended advertiser it should finish almost immediately.

So please update to the latest Zephyr (3.4.0) and re-run the tests.

> 	When the sender publishes to the receiver's unicast address with a period of 100ms, the actual publication period is around 430ms. According to the formula (10ms + (CONFIG_BT_MESH_NETWORK_TRANSMIT_INTERVAL + 10) * CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT) * N, it should be only 90ms.

Sorry, I missed that segment acknowledgement timeout part. Ack is scheduled on receiver side as soon as first segment is received and won’t be rescheduled until times out (even if another segment comes in the mean time). The timeout for acknowledgment is defined here: https://github.com/zephyrproject-rtos/zephyr/blob/c214ec3cf410ee93e519bf3b34421d1af6230af0/subsys/bluetooth/mesh/transport_legacy.c#L991-L1016, which is at least 400ms. The maximum value is defined by these parameters: CONFIG_BT_MESH_SEG_ACK_BASE_TIMEOUT, CONFIG_BT_MESH_SEG_ACK_PER_HOP_TIMEOUT, CONFIG_BT_MESH_SEG_ACK_PER_SEGMENT_TIMEOUT.

> I repeated the test, but this time publishing to a group address to avoid the acknowledgement process. According to the mesh profile (page 60), publishing to a group address should not send Segment Acknowledgment messages. However, I found that the actual publication period was around 1.8 seconds.

Correct, segment acknowledgements are not sent in this case, but the message is additionally repeated CONFIG_BT_MESH_TX_SEG_RETRANS_COUNT number of times with CONFIG_BT_MESH_TX_SEG_RETRANS_TIMEOUT_GROUP interval (which is as I see 50ms). So it should not have taken 1.8 seconds. How do you check the period time in this case? Also, please try with the latest Zephyr release.

> How can we understand these behaviors and what parameters should we change to achieve the minimum publication period and the maximum throughput allowed by Bluetooth mesh?

> How can we determine the limits of Bluetooth mesh in such cases?

The behaviour (apart from the implementation specific details) is explained in the specification. We definitely missing a guide explaining these configuration parameters and implementation details so. I think the guide should come a bit later when we finalise Bluetooth mesh 1.1 spec support.

OmarM17, 2023-07-09:
Thank you for your assistance, and I appreciate your help. I apologize for the delay in my response as I was on a holiday. 
I did some tests with the latest version, and I have achieved a minimal interval of 300ms in the receiver using a 100ms period sender. 
However, I am still in the process of testing to determine the maximum throughput. Once I have more results, I will provide additional details. Thank you again for your support.

OmarM17, 2023-07-24:
Hi,

I ran several tests, increasing the number of senders, and attempted to tune the mesh parameters. Each sender is publishing 41 bytes with a periodic publication period set to 100ms. However, after having 5 senders active for many hours, the receiver started showing the following error:

`<err> bt_mesh_transport: Out of network buffers`

I tried increasing the parameters related to buffers, but unfortunately, I got the same result.
```
CONFIG_BT_MESH_ADV_BUF_COUNT=256
CONFIG_BT_MESH_SEG_BUFS=12000
CONFIG_BT_MESH_RX_SEG_MSG_COUNT=255
```

I am wondering if I have reached the limit of the mesh network, or if there might be another parameter that I need to increase to resolve this issue. Any advice or suggestions on how to address this problem would be greatly appreciated. Thank you in advance for your help!

PavelVPV, 2023-07-24:
Hi, could you please share your current device configuration again and which address you use as target: unicast or group?

This message:
```
<err> bt_mesh_transport: Out of network buffers
```
means that the device ran out of buffers controlled by `CONFIG_BT_MESH_ADV_BUF_COUNT` option. But depending on the scenario, increasing it in may just postpone the issue.

With the following configuration, when I have just one sender (and no receivers to get max possible transmission time) the device is able to finish transmission (run out of retransmit attempts) of 41 bytes Access PDU to group address in ~98ms and ~627ms to unicast address:
```
CONFIG_BT_MESH_NETWORK_TRANSMIT_COUNT=0
CONFIG_BT_MESH_TX_SEG_RETRANS_COUNT=1
CONFIG_BT_MESH_TX_SEG_RETRANS_TIMEOUT_UNICAST=200
CONFIG_BT_MESH_TX_SEG_RETRANS_TIMEOUT_GROUP=20
```

Since you have multiple devices, they will start relaying other messages (if they have `BT_RELAY` enabled), which may delay transmission of the local message. Therefore, I recommend to use this configuration for advertiser:
```
CONFIG_BT_MESH_RELAY_RETRANSMIT_COUNT=0
CONFIG_BT_MESH_RELAY_ADV_SETS=2
CONFIG_BT_EXT_ADV_MAX_ADV_SET=3
```
In this case, the transmission of the local message may be still delayed, but not so dramatically.

If devices have Proxy feature enabled, also enable `CONFIG_BT_MESH_ADV_EXT_GATT_SEPARATE=y` and increment `CONFIG_BT_EXT_ADV_MAX_ADV_SET` by one.

OmarM17, 2023-08-08:
Thank you for your assistance.

While the warnings and errors related to Bluetooth mesh aren't causing showstoppers in my application for now, they do impact performance. I've noticed that increasing the publication period has helped the application run smoothly for now. During my latest troubleshooting efforts, I've been focusing on addressing the FATAL ERRORS.

After extensive testing, I've come to the conclusion that the FATAL ERRORS I mentioned in my previous [comment](https://github.com/zephyrproject-rtos/zephyr/issues/58455#issuecomment-1591190942) are tied to the USB sending process, not Bluetooth. In my setup, the received data is transmitted via USB to a PC running a Python code. Interestingly, when I disable the USB functionality, the FATAL ERRORS no longer occur even after running for an extended period.

Like i mentioned above when I used the addr2line command, I obtained the following results:

```
>> addr2line -e build/zephyr/zephyr.elf -a 0x0002b43c
0x0002b43c
C:/zephyrproject/3.1.99/zephyrproject/zephyr/lib/libc/minimal/source/string/string.c:111
---
>> addr2line -e build/zephyr/zephyr.elf -a 0x00027480
0x00027480
C:/zephyrproject/3.1.99/zephyrproject/zephyr/kernel/init.c:334
---
>> addr2line -e build/zephyr/zephyr.elf -a 0x0002695a
0x0002695a
C:/zephyrproject/3.1.99/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1035
---
addr2line  -e build/zephyr/zephyr.elf -a 0x20001ae8
0x20001ae8
C:\zephyrproject\3.2.99\zephyrproject\zephyr\include\zephyr\sys\dlist.h:213
```
For the USB part, I've been populating a buffer with data and using a timer to periodically send it using the usb_transfer() function. My configuration settings are as follows:

```
CONFIG_STDOUT_CONSOLE=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="testusb sample"
CONFIG_USB_DEVICE_LOOPBACK=y
```
If you have any advice regarding the simultaneous use of Bluetooth mesh and USB, I would greatly appreciate your insights. It's important for me to ensure the seamless coexistence of these two subsystems, and any recommendations or best practices you can share would be invaluable. Thank you for your assistance and expertise.

Thank you again for your support.

PavelVPV, 2023-08-21:
Could you try using gdb with enabled `CONFIG_DEBUG_THREAD_INFO` as suggested in this comment: 249199ec5a5c31d170659921048764e96d05cc0e Then you should be able to see backtrace when assertion happens.

One trick you can do with assertion in `string.c` if you still get it is to add busy loop if `s` is NULL: https://github.com/zephyrproject-rtos/zephyr/blob/4256cd41df6c60f1832fd2deb14edc30ac7debab/lib/libc/minimal/source/string/string.c#L111 This is the reason for the assertion I guess. The same you can try with `sys_dnode_init`: https://github.com/zephyrproject-rtos/zephyr/blob/07c6af3b8c35c1e49186578ca61a25c76e2fb308/include/zephyr/sys/dlist.h#L213
Then wait until device hangs and attach gdb to see the backtrace. `CONFIG_NULL_POINTER_EXCEPTION_DETECTION_MPU` option could also help to catch null pointers.

I may suggest also to try the latest zephyr version as well.

closed 2023-11-04 by github-actions[bot]