zephyrproject-rtos/zephyr #58468: Unable to pull-up spi4_sck_pe2 on STM32

yashi, 2023-05-31:
**Describe the bug**

To set `bias-pull-up` on some pins on STM32, you first need to the delete the default pull-down property.  Otherwise, the device driver happly set _both_ pull-up and pull-down to the hardware register, which is documented *reserved* and should not be used.

I can reproduced this issue on board `nucleo_f429zi` but I'm sure it is an issue on other STM32s.

**To Reproduce**
Steps to reproduce the behavior:
1. Copy the [app.overlay](https://gist.github.com/yashi/e4a492219d23d4bfbcc289b90f8f8465) to `samples/hello_world/`
2. `west build -b nucleo_f429zi samples/hello_world`
3. Look for `spi4_sck_pe2` in `build/zephyr/zephyr.dts`.  It has both `bias-pull-down` and `bias-pull-up`.
4. `west flash` and poke at `0x4002100c` to see the register has `0x3` for `PUPDR2[1:0]`, which is the pull configuration register for the pin.

**app.overlay**
```
&spi4 {
	pinctrl-0 = <&spi4_sck_pe2 &spi4_miso_pe5 &spi4_mosi_pe6>;
	pinctrl-names = "default";
	cs-gpios = <&gpioe 4 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
	status = "okay";
};

&spi4_sck_pe2 {
	bias-pull-up;
};
```

**Expected behavior**
Either:
- dts or dts compiler knows that setting both pull-up and pull-down is an error (I don't know but on some chip, it might be Ok to do so?)
- device driver checks to see the given value is OK for the hardware.

**Work around**
You have to remove the default `bias-pull-down`.
```
&spi4_sck_pe2 {
	/delete-property/ bias-pull-down;
	bias-pull-up;
};
```

**Environment (please complete the following information):**
 - OS: Linux (but doesn't matter)
 - Toolchain: Zephyr SDK (but it shouldn't matter)
 - Version tested: v3.4.0-rc1-67-g9515a5228be

**Additional context**
https://discord.com/channels/720317445772017664/884345669215735818/1112857879054135347

gmarull, 2023-05-31:
Default properties should be removed from all pin nodes in ST, as indicated in the guidelines. It’s now the only vendor left doing this, mainly for historical reasons (pre-pinctrl)

erwango, 2023-06-06:
> Default properties should be removed from all pin nodes in ST, as indicated in the guidelines. It’s now the only vendor left doing this, mainly for historical reasons (pre-pinctrl)

Well, actually this behavior is common to all vendors using pinctrl compatible based on `pincfg-node.yaml`:
If `bias-pull-down` is defined at board level (.dts), defining `bias-pull-up` in an .overlay file would end up in the same situation of having both properties defined, right ?
I don't see anything specific to STM32 pinctrl here (while I admit that it is more likely to happen since it can also happen between `-pinctrl.dtsi` and board.dts files).

gmarull, 2023-06-06:
> Well, actually this behavior is common to all vendors using pinctrl compatible based on `pincfg-node.yaml`: If `bias-pull-down` is defined at board level (.dts), defining `bias-pull-up` in an [...]

that's true, but maybe it's an option that is not required to be set at board level?

erwango, 2023-06-06:
> that's true, but maybe it's an option that is not required to be set at board level?

Let's not discuss this in current issue so that we don't lose ourselves by mixing topics.
Moving this issue to device tree bindings, since this is not STM32 specific.

erwango, 2023-06-06:
@yashi, I'm suggesting to edit topic to something like: `pintcrl: dts configuration check would be required/nice to have`
Also, I think this is rather an enhancement than a bug. We're using Linux bindings, current behavior is as expected by device tree specification.

gmarull, 2024-02-13:
Closing, this is a feature each vendor can implement as a post-build step by using the edtlib.pickle file contents.

closed 2024-02-13 by gmarull