zephyrproject-rtos/zephyr #58487: release: Make sure status of experimental and unstable APIs is updated in time for 3.4

kartben, 2023-05-31:
**Describe the bug**

Looking at https://docs.zephyrproject.org/latest/develop/api/overview.html, I get the impression that some "Experimental" APIs may effectively be "unstable" or even "stable" but just need to be promoted as such. With 3.4 approaching, it would be great to make sure this list reflects the reality of what APIs' status are.

Below is the **list of APIs currently marked as experimental or unstable**, excluding the ones added in 3.4 timeframe, and for 3.3 I took the liberty to not tag owners to not spam unnecessarily as these are probably (still) appropriately marked.

**I would suggest the maintainers for said APIs confirm that the current status is accurately stated, or step up to formally ask for promoting them to a different level of maturity.**

FWIW I am particularly interested in making sure that something like **Power Management** is properly flagged since I was really surprised to see it not even being Unstable.

/cc @nashif @carlescufi

| API | Status | Version Introduced | Maintainer(s) |
|---|---|---|---|
| Audio Codec | Experimental | 1.13 | :warning: no maintainer? |
| Digital Microphone (DMIC) | Experimental | 1.13 | ⚠️ no maintainer? |
| CoAP | Unstable | 1.10 | @rlubos |
| Counter | Unstable | 1.14 |  @nordic-krch |
| Digital-to-Analog Converter (DAC) | Experimental | 2.3 | @martinjaeger |
| Digital Audio Interface (DAI) | Experimental | 3.1 | :warning: no maintainer?<br>@juimonen @marcinszkudlinski @abonislawski |
| Display Interface | Unstable | 1.14 | @jfischer-no |
| EC Host Command | Experimental | 2.4 | @semihalf-niedzwiecki-dawid |
| Error Detection And Correction (EDAC) | Experimental | 2.5 | @finikorg |
| Fuel Gauge | Experimental | 3.3 |  |
| Gaussian &amp; Neural Accelerator (GNA) | Experimental | 1.14 | ⚠️ no maintainer? |
| I2C Target API | Experimental | 1.12 | @teburd |
| Improved Inter-Integrated Circuit (I3C) Bus | Experimental | 3.2 | @dcpleung |
| Lightweight M2M (LWM2M) | Unstable | 1.9 | @rlubos |
| LoRa and LoRaWAN | Experimental | 2.2 | @Mani-Sadhasivam |
| LoRa and LoRaWAN | Experimental | 2.5 | @Mani-Sadhasivam |
| Multi-Channel Inter-Processor Mailbox (MBOX) | Experimental | 1.0 | @carlocaione |
| MQTT | Unstable | 1.14 | @rlubos |
| MIPI Display Serial Interface (DSI) | Experimental | 3.1 | @gmarull ? |
| Pin Control API | Experimental | 3.0 | @gmarull |
| Power Management | Experimental | 1.2 | @nashif @ceolin |
| Regulators | Experimental | 2.4 | @gmarull |
| Reset Controller | Experimental | 3.1 | @andrei-edward-popa ? |
| Real Time I/O (RTIO) | Experimental | 3.2 | @teburd |
| Stream Flash | Experimental | 2.3 | ⚠️ no maintainer? |
| Secure Digital High Capacity (SDHC) | Experimental | 3.1 | @danieldegrasse |
| Task Watchdog | Experimental | 2.5 | no maintainer?<br> @katsuster @martinjaeger |
| USB Type-C Port Controller (TCPC) | Experimental | 3.1 | @sambhurst |
| UART async | Unstable | 1.14 | @dcpleung |
| USB-C device support | Experimental | 3.3 | |
| USB-C VBUS | Experimental | 3.3 | |
| Utilities | Experimental | 2.4 | I feel like this one maybe deserves<br> special treatement |
| 1-Wire Bus | Experimental | 3.2 | @str4t0m |
| Digital Signal Processing (DSP) | Experimental | 3.3 | |

henrikbrixandersen, 2023-06-06:
To be honest, I don't think this qualifies as a "bug".

finikorg, 2023-06-08:
Probably adding `Fixes` was not exactly right, reopening.

kartben, 2023-07-20:
@jhedberg do you want to piggyback this issue for 3.5.0 (looks like you might as per the milestone update) or should it be closed? Thanks!

PR #59053 (2023-06-08): doc: api: Promote EDAC API as Unstable
Promote Error Detection And Correction (EDAC) API as Unstable.

Fixes: #58487
commit 124e0f57 (2023-06-08): doc: api: Promote EDAC API as Unstable

Promote Error Detection And Correction (EDAC) API as Unstable.

Fixes: #58487
closed 2023-09-19 by nashif