zephyrproject-rtos/zephyr #58493: Not able to parse JSON with nestected objects

darshan-shah-git, 2023-05-31:
Hello,
I am developing simple app to parse JSON. That JSON is formatted as below,
`{"nodes":[{"nodeId":6,"comp":[{"length":6255},{"length":12505}]},{"nodeId":4,"comp":[{"length":505},{"length":53905}]}]}`
I already verified the above json through https://jsonlint.com/ and it is ok.

Below is the sample code that I have used,
[json_test.zip](https://github.com/zephyrproject-rtos/zephyr/files/11615686/json_test.zip)

**Commands to compile and run:** west build -b qemu_x86 . ; west build -t run

json_obj_parse is providing success but when I print the value of structure, they are different.
I debug and found that decode_num is providing correct number but it is not filling properly in structure.

henrikbrixandersen, 2023-06-01:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

mrfuchs, 2023-06-12:
You are using multidimensional arrays. Could you give #50816 a try? After your positive feedback, I'll reopen this PR and try to find enough reviewers to get it merged.

mrfuchs, 2023-06-16:
Thanks for your feedback. I reopened PR #50816.

mrfuchs, 2023-06-30:
Fixed by #50816 which was merged in ae6aa61.

closed 2023-06-30 by mrfuchs