zephyrproject-rtos/zephyr #58494: drivers: ieee802154: RSSI value converted inconsistently across drivers

fgrandel, 2023-05-31:
**Describe the bug**

The RSSI value in net_pkt (net_pkt_cb_ieee802154.rssi) is used inconsistently across drivers.

Example 1: At least one driver (nRF5) casts the signed RSSI (dBm?) value returned from the HW to the unsigned net_pkt_cb_ieee802154.rssi value without conversion.

Example 2: The cc13/26xx drivers convert the signed RSSI dBm value returned from the HW to an unsigned unitless value evenly spread to the 0-255 uint8 range of net_pkt_cb_ieee802154.rssi.

What is the intended use and range of this value? Depending on the answer one of the two implementations is probably wrong.

**To Reproduce**

n/a

**Expected behavior**

IMO, to make net_pkt_cb_ieee802154.rssi meaningful it should be a signed int8_t measured in dBm. This would "fix" the cast in the nRF5 driver, simplify other drivers and make RSSI values comparable across drivers.

I can provide fixes for net core and at least the nRF5 and CC13/26xx drivers if we agree on this intended behavior.

**Impact**

RSSI value is not comparable between different driver implementations and physically meaningless. Therefore it is probably not too useful right now at all.

**Logs and console output**

n/a

**Environment (please complete the following information):**

n/a

nandojve, 2023-05-31:
Just to mention that IEEE Std 802.15.4-2020 well define the **RSSI** or **macRSSI** and maybe we should consider this to take decision.

_6.16.2.8 Received Signal Strength Indicator (RSSI)
...
RSSI is represented as one octet of integer as shown in Table 8-108; therefore, the minimum and maximum values
are 0 (–174 dBm) and 254 (80 dBm), respectively. 255 is reserved._

There are mentions of `RSSI` at _MCPS-DATA.indication_ introduced at IEEE Std 802.15.4-2015 and `macRSSI` at _7.4.2.17 SRM IE_ since IEEE Std 802.15.4-2020.

nandojve, 2023-06-01:
There is another option which is provide a `dBm` version of set `rssi` that maps to correct integer range directly.
https://github.com/zephyrproject-rtos/zephyr/blob/941e4f000b4175b7c7d8cb30c56931b172eb68d3/include/zephyr/net/ieee802154_pkt.h#L90-L93

fgrandel, 2023-06-03:
@rlubos Just a heads up to you as I suppose that this is your area of maintenance. If you agree with the solution proposed by Gerson I'll provide a patch.

rlubos, 2023-06-05:
Sounds reasonable to standardize this. Just note, that OT layer should do the reverse conversion, as it expects RSSI in dBm.

> At least one driver (nRF5) casts the signed RSSI (dBm?)

It's most likely the case, but @jciupis can you confirm that the nRF 15.4 driver reports RSSI in dBm?

jciupis, 2023-06-06:
@rlubos that's correct, the nRF driver reports RSSI in dBm.

fgrandel, 2023-06-07:
@rlubos @nandojve @jciupis Thanks for looking into this. I'll provide a PR for this along the lines proposed by @nandojve then. :-)

PR #59133 (2023-06-17): net: l2: ieee802154: standardize RSSI value
The RSSI value in net_pkt (net_pkt_cb_ieee802154.rssi) was used inconsistently across drivers. Some drivers did cast a signed dBm value directly to net_pkt's unsigned byte value. Others were assigning the negative value of the signed dBm value and again others were offsetting and stretching the signed dBm value linearly onto the full unsigned byte range.

This change standardizes net_pkt's rssi attribute to represent RSSI on the RX path as an unsigned integer ranging from 0 (–174 dBm) to 254 (80 dBm) and lets 255 represent an "unknown RSSI" (IEEE 802 [... cut]
commit b224a099 (2023-06-17): net: l2: ieee802154: standardize RSSI value

The RSSI value in net_pkt (net_pkt_cb_ieee802154.rssi) was used
inconsistently across drivers. Some drivers did cast a signed dBm value
directly to net_pkt's unsigned byte value. Others were assigning the
negative value of the signed dBm value and again others were offsetting
and stretching the signed dBm value linearly onto the full unsigned byte
range.

This change standardizes net_pkt's rssi attribute to represent RSSI on
the RX path as an unsigned integer ranging from 0 (–174 dBm) to 254 (80
dBm) and lets 255 represent an "unknown RSSI" (IEEE 80 [... cut]
closed 2023-06-17 by cfriedt