zephyrproject-rtos/zephyr #58505: [Coverity CID: 316121] Uninitialized scalar variable in subsys/bluetooth/shell/bt.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/shell/bt.c

Category: Uninitialized variables
Function: `cmd_oob_remote`
Component: Bluetooth
CID: [316121](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316121)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/shell/bt.c#L2942

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_UNINIT). ([CWE-457](http://cwe.mitre.org/data/definitions/457.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jhedberg, 2023-06-01:
Likely false positive

jhedberg, 2023-06-01:
Duplicate of #58504

closed 2023-06-01 by jhedberg