zephyrproject-rtos/zephyr #58516: [Coverity CID: 316270] Macro compares unsigned to 0 in subsys/bluetooth/mesh/access.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/mesh/access.c

Category: Integer handling issues
Function: `count_mod_ext`
Component: Bluetooth
CID: [316270](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316270)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/mesh/access.c#L466

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_NO_EFFECT). ([CWE-570](http://cwe.mitre.org/data/definitions/570.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

PavelVPV, 2023-06-01:
Looks like false positive because `mod_rel_list` array is zero length by default.

PavelVPV, 2023-06-01:
Converted priority to low as this feature is still experimental.

closed 2023-06-05 by PavelVPV