zephyrproject-rtos/zephyr #58518: [Coverity CID: 316293] Uninitialized pointer read in subsys/bluetooth/services/ots/ots_oacp.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/services/ots/ots_oacp.c

Category: Memory - illegal accesses
Function: `bt_gatt_ots_oacp_write`
Component: Bluetooth
CID: [316293](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316293)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/services/ots/ots_oacp.c#L727

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_UNINIT). ([CWE-457](http://cwe.mitre.org/data/definitions/457.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jhedberg, 2023-06-01:
@pirun looks like this is related to your code?

pirun, 2023-06-02:
https://github.com/zephyrproject-rtos/zephyr/pull/58768

PR #58768 (2023-06-02): Bluetooth: OTS: Fix coverity issue
Fixed issue Coverity CID: 316293 https://github.com/zephyrproject-rtos/zephyr/issues/58518

Fixes #58518
closed 2023-06-02 by nashif