zephyrproject-rtos/zephyr #58552: [Coverity CID: 316583] Uninitialized scalar variable in subsys/bluetooth/shell/gatt.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/shell/gatt.c

Category: Uninitialized variables
Function: `cmd_resubscribe`
Component: Bluetooth
CID: [316583](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316583)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/bluetooth/shell/gatt.c#L629

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_UNINIT). ([CWE-457](http://cwe.mitre.org/data/definitions/457.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jhedberg, 2023-06-01:
Likely false positive

jhedberg, 2023-06-01:
Duplicate of #58504

closed 2023-06-01 by jhedberg