zephyrproject-rtos/zephyr #58565: [Coverity CID: 211475] Unintended sign extension in drivers/sensor/wsen_itds/itds.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/sensor/wsen_itds/itds.c

Category: Integer handling issues
Function: `itds_accl_channel_get`
Component: Drivers
CID: [211475](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=211475)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/sensor/wsen_itds/itds.c#L266

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_SIGN_EXTENSION). ([CWE-194](http://cwe.mitre.org/data/definitions/194.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2023-06-01:
@mah-eiSmart can you take a look?

mah-eiSmart, 2023-06-01:
> 

@MaureenHelm yes, overflow may happen here. The driver was not provided by us as a manufacturer of the product. We can provide our own implementation of it, but that will take time. However, to fix this issue the conversion can be done as floats. 

```
for (i = ofs_start; i <= ofs_stop ; i++, val++) {
  /* Sensitivity is exposed in ug/LSB */
  /* Convert to m/s^2 */
  float temp = ((float)ddata->samples[i] * (float)ddata->scale * (float)SENSOR_G) /1000000.0f);
  int64_t dval = (int64_t)temp;
  val->val1 = (int32_t)(dval / 1000000);
  val->val2 = (int32_t)(dval % 1000000);
  }
```

What's the next step? 
I do not have access to Synopsys (coverity.com) code details.

fabiobaltieri, 2023-08-02:
@mah-eiSmart this is not an overflow issue, it's a sign extension, see https://cwe.mitre.org/data/definitions/194.html.

I think the fix may just be to change scale to signed in `struct itds_device_data`, then the whole expression would use signed and should work correctly.

Could you send a PR for that?

PR #61097 (2023-08-15): drivers: sensor: correct scale in WSEN_ITDS driver
correct scale in WSEN_ITDS driver to overcome sign extension issues

Fixes #58565
closed 2023-08-15 by MaureenHelm