zephyrproject-rtos/zephyr #58566: [Coverity CID: 215224] Parse warning in drivers/modem/hl7800.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/modem/hl7800.c

Category: Parse warnings
Function: ``
Component: Drivers
CID: [215224](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=215224)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/modem/hl7800.c#L3957

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_PW.PARAM_SET_BUT_NOT_USED). ([CWE-398](http://cwe.mitre.org/data/definitions/398.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-02:
Hey @rerickson1 any chance you could look into these or reassign them? There's quite a few in modem:

https://github.com/zephyrproject-rtos/zephyr/issues/58567
https://github.com/zephyrproject-rtos/zephyr/issues/58571
https://github.com/zephyrproject-rtos/zephyr/issues/58572
https://github.com/zephyrproject-rtos/zephyr/issues/58573
https://github.com/zephyrproject-rtos/zephyr/issues/58578
https://github.com/zephyrproject-rtos/zephyr/issues/58581
https://github.com/zephyrproject-rtos/zephyr/issues/58588
https://github.com/zephyrproject-rtos/zephyr/issues/58589
https://github.com/zephyrproject-rtos/zephyr/issues/58597

rerickson1, 2023-08-02:
None of the `scan9.scan...` links work. I get the following:
[image]
I can sign in here, but how do I go about finding out what these problems are so I can fix them?
[image]

fabiobaltieri, 2023-08-02:
Good question, usually I look at the CWE link and the code and that's enough but these "Parse warning" ones are a bit of a mystery.

@nashif any idea what to do with these?

ceolin, 2024-01-31:
@rerickson1 are you able to see this CID in coverity ? you can search for the CID there. 

The problem seems to be that `sock_read` is not using `len`.

rerickson1, 2024-01-31:
> @rerickson1 are you able to see this CID in coverity ? you can search for the CID there.
> 
> The problem seems to be that `sock_read` is not using `len`.

Thanks, I will take a look.

closed 2024-02-08 by henrikbrixandersen