zephyrproject-rtos/zephyr #58574: [Coverity CID: 248434] Reliance on integer endianness in drivers/sensor/tmp108/tmp108.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/sensor/tmp108/tmp108.c

Category: Integer handling issues
Function: `tmp108_attr_get`
Component: Drivers
CID: [248434](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=248434)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/sensor/tmp108/tmp108.c#L175

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_INCOMPATIBLE_CAST). ([CWE-188](http://cwe.mitre.org/data/definitions/188.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-02:
@catch-twenty-two could you look into this?

catch-twenty-two, 2023-08-03:
Sure, I can take a look.

ceolin, 2024-01-31:
Duplicated in https://github.com/zephyrproject-rtos/zephyr/issues/67965

closed 2024-01-31 by ceolin