zephyrproject-rtos/zephyr #58577: [Coverity CID: 316178] Missing parentheses in drivers/led_strip/tlc5971.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c

Category: Incorrect expression
Function: `tlc5971_data_byte25`
Component: Drivers
CID: [316178](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316178)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L127

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-783](http://cwe.mitre.org/data/definitions/783.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

simonguinot, 2024-02-10:
According to this [comment](https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L81-L85), the bitfield of data byte 25 is composed as follow:
- bits 0-5: upper 6 bits of color 2
- bits 6-7: the 2 lower bits of color 1

And then I think that the `tlc5971_data_byte25` function should return:
`FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1) | FIELD_PREP(TLC5971_BYTE25_GBC2_MASK, gbc_color_2 >> 1)`
Instead of:
`FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1 << 6) | FIELD_PREP(TLC5971_BYTE25_GBC2_MASK, gbc_color_2 >> 1)`

Since we are interested in the two lower bits of gbc_color_1, then the left shifting is wrong.

simonguinot, 2024-02-10:
@aunsbjerg please can you have a look ?

Or is there someone with a tlc5971 controller, willing to do some testing ?

aunsbjerg, 2024-02-10:
> @aunsbjerg please can you have a look ?
> 
> Or is there someone with a tlc5971 controller, willing to do some testing ?

Will take a look in the coming week

aunsbjerg, 2024-02-12:
> According to this [comment](https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L81-L85), the bitfield of data byte 25 is composed [...]

I had to double check and I believe the shifting is still correct, but I agree that the description is not super clear. 

Each control byte has associated masks that define all bits of the byte, e.g. `TLC5971_BYTE25_GBC1_MASK` and `TLC5971_BYTE25_GBC2_MASK` for byte 25, which define the upper two and lower five bits, respectively. The confusing part is that the upper two bits of byte 25 is the _lower_ two bits of color 1. That is why we need to shift the input before we apply the mask using `FIELD_PREP`.

Without the shifting, we end up using the _upper_ two bits of color 1 value instead of the _lower_ two bits. I've tested your suggested changes, and in practice the effect is negligible as the lower bits hardly have any visible impact on the brightness.

Alternatively, we can adjust the masks and shift the output of the `FIELD_PREP` calls instead. That might make it a bit clearer?

I will add a PR with the parenthesis fixes as soon as we agree here :)

simonguinot, 2024-02-12:
> > According to this [comment](https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L81-L85), the bitfield of data byte 25 is composed as follow:
> > 
> > * bits 0-5: upper 6 bits of color 2
> > * bits 6-7: the 2 lower bits of color 1
> > 
> > And then I think that the `tlc5971_data_byte25` function should return: `FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1) | FIELD_PREP(TLC5971_BYTE25_GBC2_MASK, gbc_color_2 >> 1)` Instead of: `FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1 << 6) | FIELD_PREP(TLC5971_BYTE25_GBC2_MASK, gbc_color_2 >> 1)`
> > Since we are interested in the two lower bits of gbc_color_1, then the left shifting is wrong.
> 
> @simonguinot, I had to double check and I believe the shifting is still correct, but I agree that the description is not super clear.
> 
> Each control byte has associated masks that define all bits of the byte, e.g. `TLC5971_BYTE25_GBC1_MASK` and `TLC5971_BYTE25_GBC2_MASK` for byte 25, which define the upper two and lower five bits, respectively. The confusing part is that the upper two bits of byte 25 is the _lower_ two bits of color 1. That is why we need to shift the input before we apply the mask using `FIELD_PREP`.
> 
> Without the shifting, we end up using the _upper_ two bits of color 1 value instead of the _lower_ two bits. I've tested your suggested changes, and in practice the effect is negligible as the lower bits hardly have any visible impact on the brightness.

Hi @aunsbjerg,

First thanks for looking at this issue ! It is very much appreciated.

Now, please can you triple check :) Coverity tells us that the followings expression is always 0 whatever the value of `gbc_color_1` is:

`FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1 << 6)`

And I think it is true. Note that `FIELD_PREP` not only applies a mask but does the left shifting for you. So here you are creating 0s by left shifting 6 times `gbc_color_1`. And  FIELD_PREP shifts and masks this 0s. So the result is always 0...

So if you want to move the two lower bits of `gbc_color_1` into bits [6-7] of byte 25, then you should use:

`FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1)`

Eventually you can test this by using values in the [0-3] range for color 1. Maybe you'll see something...

aunsbjerg, 2024-02-12:
> > > According to this [comment](https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L81-L85), the bitfield of data byte 25 is composed as follow:
> > > 
> > > * bits 0-5: upper 6 bits of color 2
> > > * bits 6-7: the 2 lower bits of color 1
> > > 
> > > And then I think that the `tlc5971_data_byte25` function should return: `FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1) | FIELD_PREP(TLC5971_BYTE25_GBC2_MASK, gbc_color_2 >> 1)` Instead of: `FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1 << 6) | FIELD_PREP(TLC5971_BYTE25_GBC2_MASK, gbc_color_2 >> 1)`
> > > Since we are interested in the two lower bits of gbc_color_1, then the left shifting is wrong.
> > 
> > 
> > @simonguinot, I had to double check and I believe the shifting is still correct, but I agree that the description is not super clear.
> > Each control byte has associated masks that define all bits of the byte, e.g. `TLC5971_BYTE25_GBC1_MASK` and `TLC5971_BYTE25_GBC2_MASK` for byte 25, which define the upper two and lower five bits, respectively. The confusing part is that the upper two bits of byte 25 is the _lower_ two bits of color 1. That is why we need to shift the input before we apply the mask using `FIELD_PREP`.
> > Without the shifting, we end up using the _upper_ two bits of color 1 value instead of the _lower_ two bits. I've tested your suggested changes, and in practice the effect is negligible as the lower bits hardly have any visible impact on the brightness.
> 
> Hi @aunsbjerg,
> 
> First thanks for looking at this issue ! It is very much appreciated.
> 
> Now, please can you triple check :) Coverity tells us that the followings expression is always 0 whatever the value of `gbc_color_1` is:
> 
> `FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1 << 6)`
> 
> And I think it is true. Note that `FIELD_PREP` not only applies a mask but does the left shifting for you. So here you are creating 0s by left shifting 6 times `gbc_color_1`. And after `FIELD_PREP` shifts and masks this 0s. So the result is always 0...
> 
> So if you want to move the two lower bits of `gbc_color_1` into bits [6-7] of byte 25, then you should use:
> 
> `FIELD_PREP(TLC5971_BYTE25_GBC1_MASK, gbc_color_1)`
> 
> Eventually you can test this by using values in the [0-3] range for color 1. Maybe you'll see something...

You are correct, sorry for the inconvenience! I've opened #68864 which should fix both your comments and the parenthesis warning.

simonguinot, 2024-02-12:
Thanks @aunsbjerg !

PR #68864 (2024-02-14): drivers: led_strip: tlc5971: fix brightness control bit shifting
The lower bits of the brightness control registers were erroneously shifted even though FIELD_PREP already does the shifting. In practice, the effect is negliglible but of course should be fixed.

Also add missing parenthesis around shifting operations to satisfy static analysis.

Fixes #58577
Fixes #58590
closed 2024-02-14 by henrikbrixandersen