zephyrproject-rtos/zephyr #58587: [Coverity CID: 316354] Buffer not null terminated in drivers/wifi/esp_at/esp.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/wifi/esp_at/esp.c

Category: Memory - illegal accesses
Function: `on_cmd_cwjap`
Component: Drivers
CID: [316354](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316354)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/wifi/esp_at/esp.c#L315

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_BUFFER_SIZE). ([CWE-170](http://cwe.mitre.org/data/definitions/170.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

aescolar, 2024-02-19:
@rlubos @jukkar @str4t0m is the conclusion of https://github.com/zephyrproject-rtos/zephyr/pull/68901 then that this issue should be marked in coverity as false positive and closed here?

rlubos, 2024-02-20:
Would be best for the driver maintainer to confirm, but based on the description of the linked PR, it seems to me that you're correct.

closed 2024-07-10 by sylvioalves