zephyrproject-rtos/zephyr #58590: [Coverity CID: 316393] Missing parentheses in drivers/led_strip/tlc5971.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c

Category: Incorrect expression
Function: `tlc5971_data_byte24`
Component: Drivers
CID: [316393](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316393)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L140

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_CONSTANT_EXPRESSION_RESULT). ([CWE-783](http://cwe.mitre.org/data/definitions/783.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-02:
https://github.com/aunsbjerg could you look into this? Thanks!

simonguinot, 2024-02-10:
According to this [comment](https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L87-L91), the bitfield of data byte 24 is composed as follow:

- bits 0-6: all 7 bits of color 3
- bit 7:  lower bit of color 2

And then I think that the `tlc5971_data_byte24` function should return:

`FIELD_PREP(TLC5971_BYTE24_GBC2_MASK, gbc_color_2) | FIELD_PREP(TLC5971_BYTE24_GBC3_MASK, gbc_color_3)`

Instead of:

`FIELD_PREP(TLC5971_BYTE24_GBC2_MASK, gbc_color_2 << 7) | FIELD_PREP(TLC5971_BYTE24_GBC3_MASK, gbc_color_3)`

Since we are interested in the lower bit of `gbc_color_2`, then the left shifting is wrong.

simonguinot, 2024-02-10:
@aunsbjerg please can you have a look ?

Or is there someone with a tlc5971 controller, willing to do some testing ?

aunsbjerg, 2024-02-12:
> According to this [comment](https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/led_strip/tlc5971.c#L87-L91), the bitfield of data byte 24 is composed as follow:
> 
> * bits 0-6: all 7 bits of color 3
> * bit 7:  lower bit of color 2
> 
> And then I think that the `tlc5971_data_byte24` function should return:
> 
> `FIELD_PREP(TLC5971_BYTE24_GBC2_MASK, gbc_color_2) | FIELD_PREP(TLC5971_BYTE24_GBC3_MASK, gbc_color_3)`
> 
> Instead of:
> 
> `FIELD_PREP(TLC5971_BYTE24_GBC2_MASK, gbc_color_2 << 7) | FIELD_PREP(TLC5971_BYTE24_GBC3_MASK, gbc_color_3)`
> 
> Since we are interested in the lower bit of `gbc_color_2`, then the left shifting is wrong (i.e. upper bit of byte 24 is always set to 0).

See [comment on related issue](https://github.com/zephyrproject-rtos/zephyr/issues/58577#issuecomment-1938259891).

Will fix parenthesis warning in PR.

PR #68864 (2024-02-14): drivers: led_strip: tlc5971: fix brightness control bit shifting
The lower bits of the brightness control registers were erroneously shifted even though FIELD_PREP already does the shifting. In practice, the effect is negliglible but of course should be fixed.

Also add missing parenthesis around shifting operations to satisfy static analysis.

Fixes #58577
Fixes #58590
closed 2024-02-14 by henrikbrixandersen