zephyrproject-rtos/zephyr #58594: [Coverity CID: 316521] Unintentional integer overflow in drivers/sensor/ms5837/ms5837.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/sensor/ms5837/ms5837.c

Category: Integer handling issues
Function: `ms5837_compensate_02`
Component: Drivers
CID: [316521](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316521)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/drivers/sensor/ms5837/ms5837.c#L123

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERFLOW_BEFORE_WIDEN). ([CWE-190](http://cwe.mitre.org/data/definitions/190.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

MaureenHelm, 2023-06-01:
@marc-cpdesign can you take a look?

marc-cpdesign, 2023-06-02:
`temp_sq` is an int64 and `data->temperature` is int32 but i guess it needs to be promoted ...

Do you prefer

```
 temp_sq = (int64)(data->temperature - 2000) * (data->temperature - 2000);
```

or 

```
 temp_sq = 1LL * (data->temperature - 2000) * (data->temperature - 2000);
```

PR #60228 (2023-07-11): drivers: sensor: mx5837: address integer overflow.
Avoid integer overflow in temp_sq calculation.
Details see commit message.

Fixes #58585
Coverity-CID: 316294
Fixes #58594
Coverity-CID: 316521
commit cf29b8ca (2023-07-11): drivers: sensor: mx5837: address integer overflow.

Avoid integer overflow in temp_sq calculation.
For an analysis of the value ranges for the temp_sq calculation
of mx5837-02 see below:

calculation:
dT = adc_temperature - ((int32_t)(data->t_ref) << 8);
data->temperature = 2000 + (dT * data->tempsens) / (1ll << 23);
temp_sq = (data->temperature - 2000) * (data->temperature - 2000);

given needed storage sizes:
t_ref is uint16_t,
adc_temperature is uint24_t,
data->tempsens is uint16_t,

ranges
=> dT:                -16776960 <= dT <= 16777215         (25 bit)

=> data->temperature (TEMP):
  in [... cut]
closed 2023-07-11 by carlescufi