zephyrproject-rtos/zephyr #58631: Possible race condition in MCAN ISRs

gramsay0, 2023-05-31:
I think there might a race condition in the MCAN ISRs.
(drivers/can/can_mcan.c `can_mcan_line_0_isr`/`can_mcan_line_1_isr`).

The current flow is:
1. read ISRs
2. handle ISRs
3. clear ISRs

Generally you would clear ISRs before handling them to prevent the case where another ISR occurs during handling.

Also, the ISRs clear all events that they handle (`events`) rather than the subset of events that have actually occurred (`ir & events`), this could also lead to race conditions.

It looks like the Linux implementation clears the ISRs before handling them:
https://github.com/torvalds/linux/blob/master/drivers/net/can/m_can/m_can.c#L1086

Unless if they were level triggered, but the mcan spec mentions "edge-sensitive":
> Interrupt Register (IR)
The flags are set when one of the listed conditions is detected (edge-sensitive). The flags remain
set until the Host clears them. A flag is cleared by writing a ’1’ to the corresponding bit position.
Writing a ’0’ has no effect. A hard reset will clear the register. The configuration of IE controls
whether an interrupt is generated. The configuration of ILS controls on which interrupt line an
interrupt is signalled.

henrikbrixandersen, 2023-06-01:
@gramsay0 Well spotted, thank you for reporting this. Please test the fix in #58731 and report back.

henrikbrixandersen, 2023-06-01:
Note: This may be the root cause for #54937.

gramsay0, 2023-06-01:
> @gramsay0 Well spotted, thank you for reporting this. Please test the fix in #58731 and report back.

I've tested the changes and they are working on my AM62 board

PR #58731 (2023-06-02): drivers: can: mcan: fix IRQ acknowledge
Change the Bosch M_CAN ISRs to only acknowledge the IRQs handled in the given loop iteration (and not all IRQs supported by the given ISR) and move IRQ acknowledge to the front of the ISRs.

Fixes: #58631
commit bb513c97 (2023-06-02): drivers: can: mcan: fix IRQ acknowledge

Change the Bosch M_CAN ISRs to only acknowledge the IRQs handled in the
given loop iteration (and not all IRQs supported by the given ISR) and move
IRQ acknowledge to the front of the ISRs.

Fixes: #58631
closed 2023-06-02 by nashif