zephyrproject-rtos/zephyr #58647: [Coverity CID: 316433] Dereference after null check in subsys/net/l2/ppp/fsm.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/net/l2/ppp/fsm.c

Category: Null pointer dereferences
Function: `ppp_send_pkt`
Component: Networking
CID: [316433](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316433)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/net/l2/ppp/fsm.c#L400

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_FORWARD_NULL). ([CWE-476](http://cwe.mitre.org/data/definitions/476.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2023-06-02:
fsm cannot be null so this is false positive

nashif, 2023-06-02:
> fsm cannot be null so this is false positive

yes, but there is a check a few lines earlier

```
	if (fsm) {
		protocol = fsm->protocol;
	}

```

jukkar, 2023-06-05:
Hmm, indeed. After looking this with fresh eyes, there is a call to this function in `fsm.c:ppp_send_proto_rej()` https://github.com/zephyrproject-rtos/zephyr/blob/98810bc7a1649b4a8ccffa55176559f81512e283/subsys/net/l2/ppp/fsm.c#L1178 that sets fsm to null. But in that case the type of the message is `PPP_PROTOCOL_REJ` and we never get into the branch that the Coverity complains. So it still is false positive / intentional feature.

rlubos, 2023-06-07:
I confirm, `fsm` is NULL only when `ppp_send_pkt()` is called with `PPP_PROTOCOL_REJ`

rlubos, 2023-06-07:
Closing as per discussion above.