zephyrproject-rtos/zephyr #58677: [Coverity CID: 240685] Negative array index read in lib/os/multi_heap.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/lib/os/multi_heap.c

Category: Memory - illegal accesses
Function: `sys_multi_heap_add_heap`
Component: Other
CID: [240685](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=240685)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/lib/os/multi_heap.c#L38

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_NEGATIVE_RETURNS). ([CWE-129](http://cwe.mitre.org/data/definitions/129.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

andyross, 2023-06-01:
This is a false positive (it's an extraction sort: the code is guaranteed to select at least one item so the negative array index is impossible to reach in practice), I remember it coming around at some point in the past.  Not sure what the process is for annotating the source to suppress it.

fabiobaltieri, 2023-08-02:
> This is a false positive (it's an extraction sort: the code is guaranteed to select at least one item so the negative array index is impossible to reach in practice), I remember it coming around at some point in the past. Not sure what the process is for annotating the source to suppress it.

I think it should be this: https://community.synopsys.com/s/article/Suppressing-False-Positive-Intentional-defects, but I can't figure what tag to use.

@nashif can you help with this? I'm very surprised tha there's none of these in the code base right now so maybe it's just done differently.

ceolin, 2024-01-31:
Never properly triaged in Coverity, that is why this issue has been re-opened. Have done it based on this issue comments.