zephyrproject-rtos/zephyr #58678: [Coverity CID: 248364] Out-of-bounds write in subsys/testsuite/ztest/src/ztress.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/testsuite/ztest/src/ztress.c

Category: Memory - corruptions
Function: `ztress_execute`
Component: Other
CID: [248364](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=248364)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/testsuite/ztest/src/ztress.c#L364

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

nashif, 2023-05-31:
see #58706 (same issue in same function)

closed 2023-05-31 by nashif

nordic-krch, 2023-06-02:
False positive. Described in Coverity.