zephyrproject-rtos/zephyr #58681: [Coverity CID: 316011] Use of 32-bit time_t in lib/os/timeutil.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/lib/os/timeutil.c#L58

Category: High impact quality
Function: ``
Component: Other
CID: [316011](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316011)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/lib/os/timeutil.c#L58

```
52       time += 60LL * (tm->tm_min + 60LL * tm->tm_hour);
53       time += 86400LL * ndays;
54     
55       return time;
56     }
57     
>>>     CID 316011:  High impact quality  (Y2K38_SAFETY)
>>>     A declaration makes use of "time_t", which is defined as 32 bits wide on this platform, while a minimum safe bit width is 64.
58     time_t timeutil_timegm(const struct tm *tm)
59     {
60       int64_t time = timeutil_timegm64(tm);
61       time_t rv = (time_t)time;
62     
63       errno = 0;
```

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_Y2K38_SAFETY). ([CWE-197](http://cwe.mitre.org/data/definitions/197.html))

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

henrikbrixandersen, 2024-02-18:
Closing as this file is no longer present in `main` after commit 0b999c0943001c0faeafacbceda0c277f194bbe2. No new defects have been reported since the move. Thank you to @ceolin for noticing.

aescolar, 2024-02-19:
Actually closing :)

closed 2024-02-19 by aescolar