zephyrproject-rtos/zephyr #58703: [Coverity CID: 316336] Out-of-bounds access in subsys/shell/shell_log_backend.c

zephyrbot, 2023-05-31:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/shell/shell_log_backend.c

Category: Memory - corruptions
Function: `copy_to_pbuffer`
Component: Other
CID: [316336](https://scan9.scan.coverity.com/reports.htm#v29726/p12996/mergedDefectId=316336)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dae79cefaabf63086946a48ccca4094f26f146c8/subsys/shell/shell_log_backend.c#L148

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/reports.htm#v29271/p12996.

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

fabiobaltieri, 2023-08-02:
@jakub-uC @carlescufi could you look into this? (https://github.com/zephyrproject-rtos/zephyr/issues/58700 too)

jakub-uC, 2023-08-11:
I will fix it

PR #63906 (2023-10-13): shell: fix a memory corruption coverity issue
Added a condition to check if the size of the copied memory is a positive number.

Fixes #58700
Fixes #58703
commit 69d50db5 (2023-10-13): shell: fix a memory corruption coverity issue

Added a condition to check if the size of the copied
memory is a positive number.

Fixes #58700
Fixes #58703
closed 2023-10-13 by jhedberg