zephyrproject-rtos/zephyr #58729: tests-ci : kernel: timer: timer test Failed

hakehuang, 2023-06-01:


hakehuang, 2023-06-01:
same failure as https://github.com/zephyrproject-rtos/zephyr/issues/56589

danieldegrasse, 2023-06-01:
I can reproduce this issue on an RT1015, but only after flashing the sample- not after resetting the board using the RESET button. Since #56589 seems to be fixed, and the assertion message is different here, I'm reopening this issue.

danieldegrasse, 2023-06-05:
I've added some tracing code into the kernel in order to narrow this down, and I think I understand the failure a bit better now. The core issue is that when the `k_timer_start` call is issued, the process of scheduling the timer callback takes 2 ticks worth of cycles. So by the time the first timer callback fires, the system clock's time is 2 ticks worth of cycles ahead of the point the callback should have occurred. Because `z_timer_expiration_handler` schedules callbacks based on *when the event should have occurred*, the next timer expiration comes after only about 8 ticks worth of cycles.

For anyone looking to reproduce these results, I've pushed a branch here: https://github.com/nxp-zephyr/zephyr/tree/testbed/systick_timer_debug The tracing code is **very** rough, but it can give an idea of the events that occur to cause this failure. I have the best luck reproducing this if I flash the image once, let the test run until the first two testsuites have completed, then flash the image again (unclear why this is the case)

Key events in the log below:
- 13: `z_timer_expiration_handler` is scheduling the first timer event to occur as soon as possible (1 tick from the current time)
- 15: the kernel requests a timeout from the SysTick as soon as possible, which is now 2 ticks from the point that `z_timer_expiration_handler` scheduled the timer event, as two system ticks have elapsed
- 19: the SysTick driver is announcing progress to the kernel, 2 ticks later than `z_timer_expiration_handler` expected its scheduled event to run
- 21: `z_timer_expiration_handler` is scheduling the next timer event 10 ticks in the future from when the last timer event should have occurred, *which is 8 ticks in the future from real time*. It will then call the test's timer callback (`timer_period_fn`)
- 36: `z_timer_expiration_handler` is making the second callback to `timer_period_fn`, 8 ticks of real time after the first callback occured

| Event IDX | Type | Value | Details |
------------|------|------|-------------
0 | VAL | 0 |    Logged from SysTick init, timer has just been reset   |
1 | ANNOUNCEMENT | 1 |   first kernel announcement, announcing one tick has elapsed   |
2 | CURR_TICK | 1 |   Kernel has no events to run, has recorded first tick in `sys_clock_announce` |
3 | ELAPSED | 1 |  Kernel schedules next timeout. At this point the SysTick has rolled over once. |
4 | TIMEOUT | 2147483647 |   Kernel requests max timeout value as there are no events |
5 | LOAD| 16699999 |   SysTick is loaded with MAX_CYCLES value, equivalent to 334 ticks |
6 | VAL | 42028 |    Current value of SysTick while in `sys_clock_set_timeout`. At this point SysTick has rolled over twice  |
7 | ANNOUNCEMENT | 2 |   SysTick pending interrupt finally fires. Both rollovers are annouced to kernel as two ticks of progress  |
8 | CURR_TICK | 3 |  No events are ready, so kernel simply updates current tick count |
9 | ELAPSED | 0 |   Kernel is scheduling next timeout, no clock ticks have elapsed. |
10 | TIMEOUT | 2147483647 |   Kernel requests max timeout in `sys_clock_set_timeout` |
11 | LOAD | 16699999 |   Timeout is set to MAX_CYCLES value, equal to 334 ticks |
12 | VAL | 16680038 |  Current value of SysTick while in `sys_clock_set_timeout`. SysTick is counting down from last MAX_CYCLES load.  |
13 | ELAPSED | 283 |   The user's call to `k_timer_start` has resulted in `z_add_timeout` being called with a timeout of zero. An event will be added to the queue to occur 284 ticks from the `curr_tick` value (287 ticks in absolute time, 1 tick in the future from real time) |
14 | ELAPSED | 285 |  The kernel is checking the elapsed cycle count in `next_timeout` |
15 | TIMEOUT | 0 |   The kernel schedules a timeout in `sys_clock_set_timeout` as soon as possible |
16 | LOAD | 31075 |  Loading less than a tick's worth of cycles into SysTick to get an interrupt at next tick boundary |
17 | VAL | 2458950 | Current SysTick value in `sys_clock_set_timeout`  |
18 | USER | 57005 |  User tracing call in `jitter_drift.c`, indicating the `k_timer_start` call has returned.  |
19 | ANNOUNCEMENT | 286 |  The SysTick driver announces that 286 ticks have elapsed via `sys_clock_announce` |
20 | CURR_TICK | 287 |  The current tick value, as calculated within the event callback loop in `sys_clock_announce`. This is calculated based off the point that the event added by the call to `k_timer_start` *should have occurred*. The current system time is actually 2 ticks in the future.  |
21 | USER | 297 |  Call from `z_timer_expiration_handler`, indicating the next timer event is being scheduled for 297 ticks (absolute tick value). *This is 10 ticks in the future from when the timer event should have fired, but is really 8 ticks from the current time*. After this tracing call, `z_timer_expiration_handler` will call the timer expiration function, which will check the *actual time* using `k_cycle_get_64` |
22 | TIMEOUT | 10 |  Kernel is correctly requesting a timeout for ten ticks in the future  |
23 | LOAD | 456870 | `sys_clock_set_timeout` loads roughly 10 ticks worth of cycles  |
24 | VAL | 19023 |   Current SysTick value in `sys_clock_set_timeout`  |
25 | CURR_TICK | 289 |   The kernel's `sys_clock_announce` function has now accounted for the two extra ticks in the announcement. The `curr_tick` variable matches real system time. |
26 | ELAPSED | 1 |  Kernel is seeing how many ticks have elapsed in `next_timeout`  |
[... cut ...]
At this point the log stops, as I have added a breakpoint to the timer callback to catch the failure here. The second time callback occurred less than 8 ticks after the first callback. This can be seen via a signal analyzer trace of a GPIO pin:
[image]
The expected interval here is 1000us each, and the remaining callbacks hit this interval pretty closely. The issue only occurs on the first callback, because the kernel ends up getting the requested SysTick progress announcement via `sys_clock_announce` two ticks late, so the timer callback occurs two ticks after it should.

FRASTM, 2023-07-25:
Found similar issue when running  the **tests/kernel/timer/timer_behavior**/ on stm32 nucleo boards:
- nucleo_f429zi
- nucleo_l152re

```
START - test_jitter_drift_timer_startdelay                                                                                                                                                                  
periodic timer behavior test using explicit start with delay                                                                                                                                                
collecting time samples for approx 10 seconds                                                                                                                                                               
periodic timer samples gathered, calculating statistics                                                                                                                                                     
timer clock rate 168000000, kernel tick rate 10000                                                                                                                                                          
period duration statistics for 10000 samples (0 rollovers):                                                                                                                                                 
  expected: 1000 us,            168000.000000 cycles                                                                                                                                                        
  min:      899.755952 us,      151159 cycles                                                                                                                                                               
  max:      1000.000000 us,     168000 cycles                                                                                                                                                               
  mean:     999.989976 us,      167998.315900 cycles                                                                                                                                                        
  variance: 1.004786 us,        28359.091907 cycles                                                                                                                                                         
  stddev:   1.002390 us,        168.401581 cycles                                                                                                                                                           
timer start cycle 1708056689, end cycle 3388039848,                                                                                                                                                         
total time 9999899.755952 us, expected time 10000000.000000 us,                                                                                                                                             
expected time drift 0.000000 us, difference -100.244048 us                                                                                                                                                  
                                                                                                                                                                                                            
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:248: do_test_using: (min_us >= min_us_bound is false)                                                       
Shortest timer period too short (off by more than expected 10%)                                                                                                                                             
 FAIL - test_jitter_drift_timer_startdelay in 10.145 seconds            

```
(since sha1 76663444c1faa8618ddf0cef735b77ffffce4700 or so)
I found that reverting the https://github.com/zephyrproject-rtos/zephyr/pull/58479 on lib/libc/minimal/source/math/sqrt.c and lib/libc/minimal/source/math/sqrtf.c  --> testcase PASSED

teburd, 2023-09-21:
It's likely that the test would be better served by starting the timer in a timer itself so that it's properly tick aligned

teburd, 2023-09-26:
#63086 as a possible fix

FRASTM, 2023-10-03:
or  https://github.com/zephyrproject-rtos/zephyr/pull/63210   already merged

closed 2023-10-03 by FRASTM