zephyrproject-rtos/zephyr #58732: tests-ci : crypto: rand32: random_hw_xoshiro test Timeout

hakehuang, 2023-06-01:


hakehuang, 2023-06-01:
@dleach02

decsny, 2023-06-01:
Hi @hakehuang, I test on commit `941e4f000b4175b7c7d8cb30c56931b172eb68d3` which is the latest main at the time of testing, and I was not able to reproduce this issue. In my case, the xoshiro case of this test passed on the RT1060.

hakehuang, 2023-06-02:
@decsny Thanks a lot for checking, it is Ok in the 

```
*** Booting Zephyr OS build v3.4.0-rc1-146-gb12eb78f35ce ***
Running TESTSUITE rand32_common
===================================================================
START - test_rand32
Generating random numbers
Generating bulk fill random numbers
Generating bulk fill cryptographically secure random numbers
 PASS - test_rand32 in 0.067 seconds
===================================================================
TESTSUITE rand32_common succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [rand32_common]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.067 seconds
 - PASS - [rand32_common.test_rand32] duration = 0.067 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL

```

closed 2023-06-02 by hakehuang