zephyrproject-rtos/zephyr #58771: drivers: designware spi: uint8_t is not enough for fifo_depth

botbotty, 2023-06-02:
https://github.com/zephyrproject-rtos/zephyr/blob/b12eb78f35cecfda1e80cb37564d97b8a424f53b/drivers/spi/spi_dw.h#L35

According to https://www.synopsys.com/dw/doc.php/iip/DW_apb_ssi/4.04a/doc/DW_apb_ssi_databook.pdf:
[image]

FIFO depth can be 256, which will overflow an 8-bit unsigned int field.

henrikbrixandersen, 2023-06-03:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

botbotty, 2023-06-03:
> Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

issue edited

closed 2023-08-20 by github-actions[bot]