zephyrproject-rtos/zephyr #58828: Invalid pointer dereference in zsock_sendto when TCP connection stalls

mkaranki, 2023-06-05:
**Describe the bug**

zsock_sendto dereferences invalid pointer if an active TCP connection is stalled e.g. due to networking erorr so that TCP retransmissions are exceeded. This is seen with an MQTT over TLS application running on native_posix_64 board, and happens consistently.

The setup:   native_posix_64 board connected to internet via net-tools NAT scripts. The application does MQTT publish every 5 seconds and runs fine until the connectivity loss is simulated eg. by issuing Linux command:
```
sudo sysctl net.ipv4.conf.all.forwarding=0
```

The application has two threads: one doing publish and one doing zsock_poll and mqtt event handling. There should be no concurrent operations running however, as those are excluded by mutexes. The invalid dereferencing happens also if the upstream network interface is brought down. The point with the sysctl  is just to easilycause a situation which triggers TCP retransmits until the retry count is exceeded.

Debugger Backtrace:

```
Thread 13 "zephyr.elf" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffedffb640 (LWP 101162)]
is_condition_met (state=<synthetic pointer>, event=0x7fffedffa600) at /home/mikar/vdl200-zephyr/zephyr/include/zephyr/kernel.h:3069
3069		return sem->count;
(gdb) bt
#0  is_condition_met (state=<synthetic pointer>, event=0x7fffedffa600) at /workspace/zephyr/include/zephyr/kernel.h:3069
#1  register_events (events=events@entry=0x7fffedffa600, num_events=num_events@entry=1, poller=poller@entry=0x4c3e80 <_k_thread_obj_task_id+128>, just_check=just_check@entry=false) at /workspace/zephyr/kernel/poll.c:245
#2  0x0000000000487b7a in z_impl_k_poll (events=events@entry=0x7fffedffa600, num_events=num_events@entry=1, timeout=...) at /workspace/zephyr/kernel/poll.c:307
#3  0x000000000041c2b6 in k_poll (timeout=..., num_events=1, events=0x7fffedffa600) at /workspace/pbuild/zephyr/include/generated/syscalls/kernel.h:1410
  #4  send_check_and_wait (ctx=ctx@entry=0x4c8c40 <contexts+256>, status=-11, buf_timeout=buf_timeout@entry=5146, timeout=..., retry_timeout=retry_timeout@entry=0x7fffedffa67c)
    at /workspace/zephyr/subsys/net/lib/sockets/sockets.c:717
#5  0x000000000041cc70 in zsock_sendto_ctx (ctx=0x4c8c40 <contexts+256>, buf=0x4cf8a4 <_mbedtls_heap+17540>, len=133, flags=<optimized out>, dest_addr=0x0, addrlen=0) at /workspace/zephyr/subsys/net/lib/sockets/sockets.c:781
#6  0x000000000041cd52 in z_impl_zsock_sendto (sock=<optimized out>, buf=0x4cf8a4 <_mbedtls_heap+17540>, len=133, flags=0, dest_addr=dest_addr@entry=0x0, addrlen=addrlen@entry=0)
    at /workspace/zephyr/subsys/net/lib/sockets/sockets.c:799
#7  0x000000000041e55b in zsock_sendto (dest_addr=0x0, addrlen=0, flags=<optimized out>, len=<optimized out>, buf=<optimized out>, sock=<optimized out>) at /workspace/pbuild/zephyr/include/generated/syscalls/socket.h:258
#8  tls_tx (ctx=<optimized out>, buf=<optimized out>, len=<optimized out>) at /workspace/zephyr/subsys/net/lib/sockets/sockets_tls.c:819
#9  0x0000000000475bc1 in mbedtls_ssl_flush_output (ssl=<optimized out>) at /workspace/modules/crypto/mbedtls/library/ssl_msg.c:2039
#10 mbedtls_ssl_flush_output (ssl=0x4c73b8 <tls_contexts+120>) at /workspace/modules/crypto/mbedtls/library/ssl_msg.c:2011
#11 0x0000000000476107 in ssl_write_real (len=74, buf=0x7ffff0008250 "{\"state\":{\"reported\":{\"config\":{\"heppa\":\"kissa\"},\"applicationSchema\":{}}}}", ssl=0x4c73b8 <tls_contexts+120>)
    at /workspace/modules/crypto/mbedtls/library/ssl_msg.c:5672
#12 mbedtls_ssl_write (ssl=ssl@entry=0x4c73b8 <tls_contexts+120>, buf=0x7ffff0008250 "{\"state\":{\"reported\":{\"config\":{\"heppa\":\"kissa\"},\"applicationSchema\":{}}}}", len=<optimized out>)
    at /workspace/modules/crypto/mbedtls/library/ssl_msg.c:5711
#13 0x000000000041f5c1 in send_tls (flags=<optimized out>, len=<optimized out>, buf=<optimized out>, ctx=0x4c7340 <tls_contexts>) at /workspace/zephyr/subsys/net/lib/sockets/sockets_tls.c:1909
#14 ztls_sendto_ctx (ctx=0x4c7340 <tls_contexts>, buf=<optimized out>, len=<optimized out>, flags=<optimized out>, dest_addr=<optimized out>, addrlen=<optimized out>)
    at /workspace/zephyr/subsys/net/lib/sockets/sockets_tls.c:2014
#15 0x000000000041f69e in ztls_sendmsg_ctx (flags=<optimized out>, msg=<optimized out>, ctx=<optimized out>) at /workspace/zephyr/subsys/net/lib/sockets/sockets_tls.c:2064
#16 ztls_sendmsg_ctx (ctx=0x4c7340 <tls_contexts>, msg=0x7fffedffa878, flags=<optimized out>) at /workspace/zephyr/subsys/net/lib/sockets/sockets_tls.c:2031
#17 0x000000000041cee8 in z_impl_zsock_sendmsg (sock=<optimized out>, msg=msg@entry=0x7fffedffa878, flags=flags@entry=0) at /workspace/zephyr/subsys/net/lib/sockets/sockets.c:873
#18 0x000000000043d4cc in zsock_sendmsg (flags=0, msg=0x7fffedffa878, sock=<optimized out>) at /workspace/pbuild/zephyr/include/generated/syscalls/socket.h:283
#19 mqtt_client_tls_write_msg (client=0x7ffff0002408, message=0x7fffedffa878) at /workspace/zephyr/subsys/net/lib/mqtt/mqtt_transport_socket_tls.c:141
#20 0x000000000043ce08 in client_write_msg (message=0x7fffedffa878, client=0x7ffff0002408) at /workspace/zephyr/subsys/net/lib/mqtt/mqtt.c:154
#21 mqtt_publish (client=0x7ffff0002408, param=0x7fffedffa908) at /workspace/zephyr/subsys/net/lib/mqtt/mqtt.c:276
#22 0x0000000000405d65 in AwsIotZephyr2::publish_mqtt_message (this=this@entry=0x7ffff0002360, topic="$aws/things/X/shadow/update", payload=std::vector of length 74, capacity 74 = {...}, packet_id=@0x7fffedffa9a6: 20)
    at /workspace/proj/app/src/app.cpp:307
....
$1 = (struct k_poll_event *) 0x7fffedffa600
(gdb) p event->sem
$2 = (struct k_sem *) 0x90
```

The beef there is that the k_poll is trying to read the semaphore count, but the semaphore is not anymore available.

The logs ending to the crash:
```
[00:00:23.400,000] <inf> main: PUBLISH packet id: 22903, qos: 0, payload len: 278, rc: 278
[00:00:23.620,000] <dbg> net_tcp: tcp_resend_data: (tcp_work): send_data_retries=0
[00:00:23.620,000] <dbg> net_tcp: tcp_out_ext: (tcp_work): ACK,PSH Seq=3806123100 Ack=2311191188 Len=69
[00:00:23.620,000] <dbg> net_tcp: tcp_send_process_no_lock: (tcp_work): ACK,PSH Seq=3806123100 Ack=2311191188 Len=69
[00:00:23.620,000] <dbg> net_tcp: tcp_send: (tcp_work): ACK,PSH Seq=3806123100 Ack=2311191188 Len=69
[00:00:23.620,000] <dbg> net_tcp: tcp_send_data: (tcp_work): conn: 0x4fe418 total=69, unacked_len=69, send_win=2730, mss=1360
[00:00:23.620,000] <dbg> net_tcp: tcp_send_data: (tcp_work): conn: 0x4fe418 send_data_timer=0, send_data_retries=0
[00:00:23.640,000] <dbg> net_conn: net_conn_input: (rx_q[0]): Check TCP listener for pkt 0x4fc7c8 src port 8883 dst port 44539 family 1
[00:00:23.640,000] <dbg> net_conn: net_conn_input: (rx_q[0]): [0x4c9310] match found cb 0x437c40 ud 0x4c8c40 rank 0x3f
[00:00:23.640,000] <dbg> net_tcp: tcp_in: (rx_q[0]): ACK Seq=2311191188 Ack=3806123169 Len=0 [ESTABLISHED Seq=3806123100 Ack=2311191188]
[00:00:23.640,000] <dbg> net_tcp: tcp_in: (rx_q[0]): Lowering send window from 65142 to 2730
[00:00:23.640,000] <dbg> net_tcp: tcp_window_full: (rx_q[0]): conn: 0x4fe418 window_full=0
[... 235 lines cut ...]
[00:00:48.580,000] <dbg> net_tcp: tcp_resend_data: (tcp_work): send_data_retries=9
[00:00:48.580,000] <dbg> net_tcp: tcp_resend_data: (tcp_work): conn: 0x4fe418 close, data retransmissions exceeded
[00:00:48.580,000] <dbg> net_tcp: tcp_conn_close_debug: (tcp_work): conn: 0x4fe418 closed by TCP stack (tcp_resend_data():1331)
[00:00:48.580,000] <dbg> net_tcp: tcp_conn_unref: (tcp_work): conn: 0x4fe418, ref_count=1
[00:00:48.580,000] <dbg> net_conn: net_conn_unregister: (tcp_work): Connection handler 0x4c9310 removed
```

It looks like the TCP connection context is freed too early without considering there's still k_poll ongoing by the application thread, which is needing the tcp connection context to read the semaphore?

This seems a regression: if I revert commits 9fef0c521d155a26311c0e10ca0bb33df36bc0c8 and 79461a378c838c1619fdaf0fcb7c9c6a1a5f2bad then the crash does not happen and the mqtt_publish functions returns error correctly. The commits were introduced by the PR https://github.com/zephyrproject-rtos/zephyr/pull/53891 due to issue https://github.com/zephyrproject-rtos/zephyr/issues/53500

I can't share the application right now but hope the traces are giving a clue what is wrong. This  could perhaps be reproduced with the MQTT sample application. Please let me know if traces are not enough and precise way to reproduce this is needed.

**Expected behavior**

zsock_sendto should return an error which should be propagated to the application if the connection fails.

**Impact**

Poor resilience on network failures.

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA or Version used: Zephyr 3.3.0.

mkaranki, 2023-06-05:
> For a quick check if it helps, adding the state change should be okayish, 

Good news. It seems this is the case. After adding `conn_state(conn, TCP_CLOSED)` to `tcp_conn_close` there's no crash and mqtt_publish fails correctly with -EIO (-5).  The logs (likely just for filling up Github's  storage space just bit more?):

```
[00:03:16.450,000] <dbg> net_tcp: tcp_resend_data: (tcp_work): send_data_retries=9
[00:03:16.450,000] <dbg> net_tcp: tcp_resend_data: (tcp_work): conn: 0x4fe418 close, data retransmissions exceeded
[00:03:16.450,000] <dbg> net_tcp: tcp_conn_close_debug: (tcp_work): conn: 0x4fe418 closed by TCP stack (tcp_resend_data():1286)
[00:03:16.450,000] <dbg> net_tcp: tcp_conn_close_debug: (tcp_work): ESTABLISHED->CLOSED
[00:03:16.450,000] <dbg> net_tcp: tcp_conn_unref: (tcp_work): conn: 0x4fe418, ref_count=2
[00:03:16.480,000] <err> net_mqtt: Transport write failed, err_code = -5, closing connection
[00:03:16.480,000] <inf> net_mqtt_sock_tls: Closing socket 0
[00:03:16.480,000] <dbg> net_tcp: net_tcp_recv: (task_id): context: 0x4c8c40, cb: (nil), user_data: (nil)
[00:03:16.480,000] <dbg> net_tcp: net_tcp_put: (task_id):  [CLOSED Seq=1850394098 Ack=129040754]
[00:03:16.480,000] <dbg> net_tcp: net_tcp_put: (task_id): context 0x4c8c40 CONNECTED
[00:03:16.480,000] <dbg> net_tcp: tcp_conn_unref: (task_id): conn: 0x4fe418, ref_count=1
[00:03:16.480,000] <dbg> net_conn: net_conn_unregister: (task_id): Connection handler 0x4c9310 removed
[00:03:16.480,000] <err> main: mqtt_publish failed, rc:-5
[00:03:16.480,000] <inf> main: MQTT client disconnected -5
[00:03:16.480,000] <inf> main: Connection state changed to 1
```

rlubos, 2023-06-06:
@mkaranki I've submitted https://github.com/zephyrproject-rtos/zephyr/pull/58905 with a fix proposal.

PR #58905 (2023-06-06): net: tcp: Fix possible double TCP context dereferencing
In case TCP connection is being closed from the TCP stack, due to for instance retransmission timeout, the stack should also switch the TCP state to CLOSED. Otherwise, there was a risk of dereferencing the TCP context twice, for example if the application was in active socket send(), and tried to reschedule data transmission.

Additionally, make sure that the TCP_CLOSED state handling is a no-op state - otherwise, there is a risk that if packets keep incoming before the application dereferences the TCP context on its side, TCP stack will  [... cut]
closed 2023-06-06 by nashif