zephyrproject-rtos/zephyr #58860: RFC: Variants of DT_REG_ADDR() with explicit 'ranges' mapping

57300, 2023-06-05:
# Introduction

This is a proposal to introduce new cousins of the `DT_REG_ADDR()` macro, which would:

- Lookup a node's address within a particular address space.
- Distinguish between ranges-translated and untranslated addresses in code.
- When writing or refactoring DTS and overlays, help catch mistakes at compile time.
- Bring the usage of `reg` and `ranges` properties closer in line with Linux.

I've come up with a few possible approaches and I would appreciate some discussion on what would be the optimal solution.

## Motivating problem

Consider these simplified DTS fragments:
```
/* soc-xxx.dts */

/ {
	soc: soc {
		ranges;

		peripheral: peripheral@50000000 {
			reg = < 0x50000000 0x10000000 >;
			ranges = < 0x0 0x50000000 0x10000000 >;

			flash_controller: flash-controller@60000 {
				reg = < 0x60000 0x1000 >;

				flash: flash@10000000 {
					reg = < 0x10000000 0x20000 >;
				};
			};
		};

		sram: sram@30000000 {
			reg = < 0x30000000 0x10000 >;
			ranges = < 0x0 0x30000000 0x10000 >;
		};
	};
};

#include "soc-common.dtsi"
```
```
/* soc-common.dtsi */

&sram {
	shmem: shmem@e000 {
		reg < 0xe000 0x2000 >;
	};
};
```

Using Zephyr's devicetree API today, we can expect to get these node addresses:
```
DT_REG_ADDR(DT_NODELABEL(peripheral))       == 0x50000000 /* mapped to root via 'soc' */
DT_REG_ADDR(DT_NODELABEL(flash_controller)) == 0x50060000 /* mapped to root via 'soc', 'peripheral' */
DT_REG_ADDR(DT_NODELABEL(flash))            == 0x10000000 /* mapped to 'flash_controller' */
DT_REG_ADDR(DT_NODELABEL(sram))             == 0x30000000 /* mapped to root via 'soc' */
DT_REG_ADDR(DT_NODELABEL(shmem))            == 0x3000e000 /* mapped to root via 'soc', 'sram' */
```

For each node, the adjacent comment indicates which address space it is mapped to, through intermediate ranges-translations.

Now, suppose that we either modify `soc-xxx.dts`, or add `soc-yyy.dts`, with the following change:
```
&sram {
	ranges = < 0x0 0x30000000 0x8000 >;
};
```
[... 70 lines cut ...]
- `DT_REG_ADDR_BY_IDX()`
- `DT_REG_ADDR_BY_NAME()`
- `DT_INST_REG_ADDR()`
- `DT_INST_REG_ADDR_BY_IDX()`
- `DT_INST_REG_ADDR_BY_NAME()`

Changes to edtlib would encompass:

1. Turning the internal `_translate()` function into a public API: `translate_one(addr: int, node: dtlib_Node) -> Optional[int]`. This would translate `addr` to the address space of `node` and return either a valid address or `None`. It would be called iteratively, taking the previous result and mapping it onto `node.parent`. In particular, this would be suitable for generating all possible values for `DT_REG_REL_ADDR()`.
2. Adding a `Node.raw_regs` property to the public API, which would contain addresses with no translation. There is already `Node.regs`, but its values are translated in the manner of `DT_REG_ADDR()`, and it probably shouldn't be broken. Both properties would be set in `_init_regs()`.

## Impact

Initially, the rest of Zephyr will not be affected by this change. Eventually, I do hope that other developers can benefit from using the new API instead of `DT_REG_ADDR()` in their code, so that they can be more conscious of how certain addresses should be interpreted. This could result in more readable code that is also less susceptible to the type of mistakes outlined above.

# Alternatives

I've considered that one could use `DT_PROP_BY_IDX(node_id, reg, 0)` in place of `DT_REG_RAW_ADDR(node_id)`, but only for 32-bit addresses. This could be generalized by adding a different API for retrieving `#address-cells` and `#size-cells`, but I think that would be a bit clunky.

I've also thought about using the existing `DT_RANGES_*` macros to translate and validate addresses manually. This would still require at least the following additions:
- `DT_REG_RAW_ADDR()`
- Something like `DT_FOREACH_ANCESTOR()` for walking up the tree.
- A small bugfix: https://github.com/zephyrproject-rtos/zephyr/pull/58858

With this method, compile-time validation would probably result in a pretty gnarly-looking `BUILD_ASSERT()`. Runtime validation could be more reasonable, but I wouldn't consider that to be a desirable solution.

mbolivar-ampere, 2023-09-07:
I have some problems of my own that I think would be addressed by this general approach as well. Sorry I did not have time to review in detail but I support the general idea as outlined in the overview and expect to have more time on Friday to look at the details. Sorry for the delay.

mbolivar-ampere, 2023-09-13:
OK, I've read and thought about this and I think it makes perfect sense and you've got a great proposal.

I want to make a couple of minor suggestions

- `DT_REG_RAW_ADDR(node_id)`: suggest renaming to `DT_REG_ADDR_RAW`, so that we can still grep for DT_REG_ADDR to search for address lookups. This is useful when digging around in drivers, for example.
- `DT_REG_ABS_ADDR(node_id)`: "absolute" feels like a concept that will not age well. I am open to suggestions but here are a few ideas:

  - `DT_REG_ADDR_CPU`: makes it clearer that this is a physical CPU address
  - `DT_REG_ADDR_CPUS` if you're worried about clarity on SMP (I am not)
  - `DT_REG_ADDR_ROOT`: makes it clearer that this is requesting a register address in the address space of the root node (which is implicitly what /cpus looks at)
- additionally have edtlib error out if a translated register address or size won't fit in the root node's `#address-cells` or `#size-cells`, to avoid situations where errant ranges push node addresses outside of 32 bits on a 32-bit CPU, for example

rruuaanng, 2024-11-11:
I'll think about it. Maybe soon :)

rruuaanng, 2024-11-11:
> It seems to have been added, I will close this issue.
https://docs.zephyrproject.org/apidoc/latest/group__devicetree-reg-prop.html#gac6d8279c32351ced4c0ac7f32270974e

Sorry, it seems it is not implemented. Please ignore what I said.

rruuaanng, 2024-11-11:
They already exist in zephyr. So I closed them.

[image]

57300, 2024-11-11:
> They already exist in zephyr. So I closed them.

No they aren't. The macro `DT_REG_ADDR_RAW` was taken in #78766 for a different purpose than described in this RFC.
Now, to implement a macro for an address with no ranges-translation, another macro name needs to be chosen.

rruuaanng, 2024-11-11:
Sorry, maybe I didn't notice. If there are no other questions, I will implement it （I don't seem to understand what RFC needs. Maybe you can give a more accurate description. For example, the macro name you expect.）