zephyrproject-rtos/zephyr #58878: Open Charge Point Protocol (OCPP) native implementation

parthitce, 2023-06-06:
**Is your feature request related to a problem? Please describe.**
Add support for Open Charge Point Protocol (OCPP) application protocol as native implementation in zephyr.

**What is OCPP?**
- Communication protocol between the Charge Point for Electric Vehicle and the Central system (Frontend + Backend platforms like [1]).
- This OCPP application protocol typically works over Websockets on top of TCP.
- Communications by itself is represented using messages which can be either JSON or SOAP format.
- There are various specification versions available from 1.0 to 2.0.1 (latest). Please refer to [2].
- Version 2.0, 2.0.1 aren't backward compatible with 1.6.
- AFAIK the functionality is concerned, it's categorized as profiles.

1. Core profiles
2. Firmware Management
3. Smart charging
4. Reservation
5. Local Authorization List Management
6. Remote trigger

**Describe the solution you'd like**
Implement a native version for zephyr with all the profile messages implemented. AFAIK the specification is concerned, Core profile SHOULD be supported and all the remaining can be left optional. Central system will negotiate the feature support during the initial communication with the Charge Point.

- As zephyr already supports a websocket which can be consumed on top of a TCP socket, communication payload offload SHALL be consumed with zephyr's websocket API.
- Define the role of the user/application with API's which must be exposed for OCPP.
- User/Application SHALL not know any details about OCPP itself and must be abstracted, typically as a application library layer.

**Additional context**
- @ssekar15 from @linumiz already started with the design and implementation for OCPP 1.6 version of the specification.
- This feature request is opened as a placeholder to discuss the API's and sooner with a PR for zephyr
- Initial idea is to keep the scope to OCPP 1.6 version with only Core profiles with possibly the support for Local Authorization List Management

[1]: https://github.com/steve-community/steve
[2]: https://www.openchargealliance.org/downloads/

parthitce, 2023-12-26:
Current working base is posted here: https://github.com/linumiz/zephyr/pull/13, which includes,
- core profile
- consumer API for OCPP
- test sample with steve server

kartben, 2024-01-22:
@parthitce any reason why you're not making this an actual PR against Zephyr (note that your current PR seems to be modifying the hello world sample while I think you'd want a dedicated sample for OCPP...)? This might help put more eyes on this work. Might also be worthwhile for you to join our [Discord channel](https://chat.zephyrproject.org/) if you haven't already.

parthitce, 2024-02-08:
> @parthitce any reason why you're not making this an actual PR against Zephyr (note that your current PR seems to be modifying the hello world sample while I think you'd want a dedicated sample for [...]

Here is the PR with draft state: https://github.com/zephyrproject-rtos/zephyr/pull/68739