zephyrproject-rtos/zephyr #58882: Establish a way to trace existing testcases back to design specification

tobiaskaestner, 2023-06-06:
Even where design specifications and API documentation and Test Cases exist, the former are not related to the latter in an easily extractable way. 

Some test cases already contain doxygen comments, e.g. test_priority_scheduling.c but this is not extracted in any way into a meaningful documentation. 

So as a first step we might add doxygen to more test cases and then add test case references to the doxygen comments for API calls. Backward traces from the test cases to the API calls can then be added, too. 

As an PoC we might focus on the mutex API 
https://docs.zephyrproject.org/latest/kernel/services/synchronization/mutexes.html#id10
and related test cases
https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/kernel/mutex

simhein, 2023-06-19:
@tobiaskaestner sorry for the late response this was on my agenda to add the information about the PoC of a traceability mechanism with doxygen here. The PoC can be found here: https://github.com/simhein/traceability

quarterbit, 2024-01-30:
Just an idea: You may want to have a look at this project: [](https://github.com/itsallcode/openfasttrace )
It is used in automotive for requirements tracing along all artifacts from requirement to desing and tests.

simhein, 2024-01-30:
> Just an idea: You may want to have a look at this project: [OpenFastTrace](https://github.com/itsallcode/openfasttrace) It is used in automotive for requirements tracing along all artifacts from requirement to desing and tests.

Thanks for sharing. I will definitely look into it.