zephyrproject-rtos/zephyr #58892: Not able to build zephyr example application

Kannannnnn, 2023-06-06:
steps to reproduce the error
step 1) west init -m https://github.com/zephyrproject-rtos/example-application --mr main my-workspace
Output:nan@Kannan:~$ west init -m https://github.com/zephyrproject-rtos/example-application --mr main my-workspace
=== Initializing in /home/kannan/my-workspace
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/example-application, rev. main
Cloning into '/home/kannan/my-workspace/.west/manifest-tmp'...
remote: Enumerating objects: 371, done.
remote: Counting objects: 100% (296/296), done.
remote: Compressing objects: 100% (144/144), done.
remote: Total 371 (delta 127), reused 209 (delta 98), pack-reused 75
Receiving objects: 100% (371/371), 58.86 KiB | 369.00 KiB/s, done.
Resolving deltas: 100% (139/139), done.
--- setting manifest.path to example-application
=== Initialized. Now run "west update" inside /home/kannan/my-workspace.
step 2)west update
output:nan@Kannan:~$ cd my-workspace
kannan@Kannan:~/my-workspace$ west update
=== updating zephyr (zephyr):
--- zephyr: initializing
Initialized empty Git repository in /home/kannan/my-workspace/zephyr/.git/
--- zephyr: fetching, need revision main
remote: Enumerating objects: 867340, done.
remote: Counting objects: 100% (787/787), done.
remote: Compressing objects: 100% (413/413), done.
remote: Total 867340 (delta 369), reused 632 (delta 346), pack-reused 866553
Receiving objects: 100% (867340/867340), 573.63 MiB | 3.45 MiB/s, done.
Resolving deltas: 100% (657283/657283), done.
From https://github.com/zephyrproject-rtos/zephyr
 * branch                  main           -> FETCH_HEAD
 * [new tag]               v1.0.0         -> v1.0.0
 * [new tag]               v1.1.0         -> v1.1.0
 * [new tag]               v1.1.0-rc1     -> v1.1.0-rc1
 * [new tag]               v1.10.0        -> v1.10.0
 * [new tag]               v1.10.0-rc1    -> v1.10.0-rc1
 * [new tag]               v1.10.0-rc2    -> v1.10.0-rc2
 * [new tag]               v1.10.0-rc3    -> v1.10.0-rc3
 * [new tag]               v1.11.0        -> v1.11.0
 * [new tag]               v1.11.0-rc1    -> v1.11.0-rc1
 * [new tag]               v1.11.0-rc2    -> v1.11.0-rc2
 * [new tag]               v1.11.0-rc3    -> v1.11.0-rc3
 * [new tag]               v1.12.0        -> v1.12.0
 * [new tag]               v1.12.0-rc1    -> v1.12.0-rc1
 * [new tag]               v1.12.0-rc2    -> v1.12.0-rc2
 * [new tag]               v1.12.0-rc3    -> v1.12.0-rc3
 * [new tag]               v1.13.0        -> v1.13.0
 * [new tag]               v1.13.0-rc1    -> v1.13.0-rc1
 * [new tag]               v1.13.0-rc2    -> v1.13.0-rc2
 * [new tag]               v1.13.0-rc3    -> v1.13.0-rc3
 * [new tag]               v1.14.0        -> v1.14.0
 * [new tag]               v1.14.0-rc1    -> v1.14.0-rc1
 * [new tag]               v1.14.0-rc2    -> v1.14.0-rc2
 * [new tag]               v1.14.0-rc3    -> v1.14.0-rc3
 * [new tag]               v1.14.1        -> v1.14.1
 * [new tag]               v1.14.1-rc1    -> v1.14.1-rc1
 * [new tag]               v1.14.1-rc2    -> v1.14.1-rc2
 * [new tag]               v1.14.1-rc3    -> v1.14.1-rc3
 * [new tag]               v1.14.2        -> v1.14.2
 * [new tag]               v1.14.3        -> v1.14.3
 * [new tag]               v1.14.3-rc1    -> v1.14.3-rc1
 * [new tag]               v1.14.3-rc2    -> v1.14.3-rc2
 * [new tag]               v1.2.0         -> v1.2.0
 * [new tag]               v1.2.0-rc1     -> v1.2.0-rc1
 * [new tag]               v1.2.0-rc2     -> v1.2.0-rc2
 * [new tag]               v1.3.0         -> v1.3.0
 * [new tag]               v1.3.0-rc1     -> v1.3.0-rc1
 * [new tag]               v1.3.0-rc2     -> v1.3.0-rc2
 * [new tag]               v1.4.0         -> v1.4.0
 * [new tag]               v1.4.0-rc1     -> v1.4.0-rc1
 * [new tag]               v1.4.0-rc2     -> v1.4.0-rc2
 * [new tag]               v1.4.0-rc3     -> v1.4.0-rc3
[... 193 lines cut ...]
-- Board: esp32
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/home/kannan/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (/home/kannan/zephyr-sdk-0.16.1)
-- Found Dtc: /home/kannan/zephyr-sdk-0.16.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/kannan/my-workspace/zephyr/boards/xtensa/esp32/esp32.dts
In file included from /home/kannan/my-workspace/zephyr/boards/xtensa/esp32/esp32.dts:8,
                 from <command-line>:
/home/kannan/my-workspace/zephyr/dts/xtensa/espressif/esp32.dtsi:13:10: fatal error: dt-bindings/pinctrl/esp32-pinctrl.h: No such file or directory
   13 | #include <dt-bindings/pinctrl/esp32-pinctrl.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMake Error at /home/kannan/my-workspace/zephyr/cmake/modules/extensions.cmake:3714 (message):
  failed to preprocess devicetree files (error code 1):
  /home/kannan/my-workspace/zephyr/boards/xtensa/esp32/esp32.dts
Call Stack (most recent call first):
  /home/kannan/my-workspace/zephyr/cmake/modules/dts.cmake:223 (zephyr_dt_preprocess)
  /home/kannan/my-workspace/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
  /home/kannan/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/kannan/my-workspace/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/kannan/my-workspace/build -GNinja -DBOARD=esp32 -S/home/kannan/my-workspace/example-application/app
Imapact: Not able to proceed further.

carlescufi, 2023-06-06:
Did you forget to run `west update` ?

closed 2023-06-06 by carlescufi

Ananyaaynana, 2023-12-11:
I have the same issue. Ran west update after going into the my-workspace directory right after west init -m https://github.com/zephyrproject-rtos/example-application --mr main my-workspace. Not able to build the example application.