zephyrproject-rtos/zephyr #58896: Bluetooth: Controller: LE Connected Isochronous Stream Established failure 0x20

Thalley, 2023-06-06:
**Describe the bug**
Attempting to setup a bidirectional CIS to 2 peripherals 2 times seems to fail the second time. It works fine when there is only 1 peripheral, but when there are 2 peripherals it fails with `Unsupported LMP Parameter Value / Unsupported LL Parameter Value (0x20)`.

**To Reproduce**
Steps to reproduce the behavior:
Build and run the Bluetooth Audio shell (the BT shell with `audio.conf`). 

On the 2 peripherals do
```
bt init sync
bap init
cap_acceptor init
bt adv-create conn-nscan ext-adv name-ad
bt adv-data discov
bt adv-start 
```

On the central do
```
bt init
bap init
bt connect-name shell
bap discover
cap_initiator discover

bt connect-name shell
bt select <addr of the other peripheral)
bap discover
cap_initiator discover

cap_initiator unicast-start conns 2
cap_initiator unicast-stop
cap_initiator unicast-start conns 2
```
where the last command will trigger the issue. 

**Expected behavior**
Expect that it is possible to setup streams and tear them down and then set them up again with the same parameters without issue. 

**Impact**
Significant annoyance, but can be circumvented by disconnecting the ACLs inbetween. 

**Logs and console output**
[central.txt](https://github.com/zephyrproject-rtos/zephyr/files/11664450/central.txt)
[peripheral_1.txt](https://github.com/zephyrproject-rtos/zephyr/files/11664451/peripheral_1.txt)
[peripheral_2.txt](https://github.com/zephyrproject-rtos/zephyr/files/11664452/peripheral_2.txt)

The issue can be seen in the `central.txt` BTMON log where the second attempt to connect the first CIS ends up with 
```
> HCI Event: LE Meta Event (0x3e) plen 29                                                                                                   #2855 83.683674
      LE Connected Isochronous Stream Established (0x19)
        Status: Unsupported LMP Parameter Value / Unsupported LL Parameter Value (0x20)
        Connection Handle: 6
        CIG Synchronization Delay: 4440 us (0x001158)
        CIS Synchronization Delay: 4440 us (0x001158)
        Central to Peripheral Latency: 4440 us (0x001158)
        Peripheral to Central Latency: 4440 us (0x001158)
        Central to Peripheral PHY: LE 2M (0x02)
        Peripheral to Central PHY: LE 2M (0x02)
        Number of Subevents: 3
        Central to Peripheral Burst Number: 1
        Peripheral to Central Burst Number: 1
        Central to Peripheral Flush Timeout: 1
        Peripheral to Central Flush Timeout: 1
        Central to Peripheral MTU: 40
        Peripheral to Central MTU: 40
        ISO Interval: 8
```

Things I have noted from the log:
The first group is created as 
```
< HCI Command: LE Set Connected Isochronous Group Parameters (0x08|0x0062) plen 33                                                          #2446 74.956508
        CIG ID: 0x00
        Central to Peripheral SDU Interval: 10000 us (0x002710)
        Peripheral to Central SDU Interval: 10000 us (0x002710)
        SCA: 201 - 500 ppm (0x00)
        Packing: Sequential (0x00)
        Framing: Unframed (0x00)
        Central to Peripheral Maximum Latency: 10 ms (0x000a)
        Peripheral to Central Maximum Latency: 10 ms (0x000a)
        Number of CIS: 2
        CIS ID: 0x00
        Central to Peripheral Maximum SDU Size: 40
        Peripheral to Central Maximum SDU Size: 40
        Central to Peripheral PHY: LE 2M (0x02)
        Peripheral to Central PHY: LE 2M (0x02)
        Central to Peripheral Retransmission attempts: 0x02
        Peripheral to Central Retransmission attempts: 0x02
        CIS ID: 0x01
        Central to Peripheral Maximum SDU Size: 40
        Peripheral to Central Maximum SDU Size: 40
        Central to Peripheral PHY: LE 2M (0x02)
        Peripheral to Central PHY: LE 2M (0x02)
        Central to Peripheral Retransmission attempts: 0x02
        Peripheral to Central Retransmission attempts: 0x02
> HCI Event: Command Complete (0x0e) plen 10                                                                                                #2447 75.024486
      LE Set Connected Isochronous Group Parameters (0x08|0x0062) ncmd 1
        Status: Success (0x00)
        CIG ID: 0x00
        Number of Handles: 2
        Connection Handle #0: 6
        Connection Handle #1: 7
```
where the first CIS gets handle 6, and the second CIS gets handle 7. 

The first CIS are then connected as 
```
< HCI Command: LE Create Connected Isochronous Stream (0x08|0x0064) plen 5                                                                  #2570 75.690162
        Number of CIS: 1
        CIS Handle: 6
        ACL Handle: 0
```

and the second as 
```
< HCI Command: LE Create Connected Isochronous Stream (0x08|0x0064) plen 5                                                                  #2596 76.137561
        Number of CIS: 1
        CIS Handle: 7
        ACL Handle: 1
```

which works. 

On the second attempt, the group is created as 
```
< HCI Command: LE Set Connected Isochronous Group Parameters (0x08|0x0062) plen 33                                                          #2698 82.469154
        CIG ID: 0x00
        Central to Peripheral SDU Interval: 10000 us (0x002710)
        Peripheral to Central SDU Interval: 10000 us (0x002710)
        SCA: 201 - 500 ppm (0x00)
        Packing: Sequential (0x00)
        Framing: Unframed (0x00)
        Central to Peripheral Maximum Latency: 10 ms (0x000a)
        Peripheral to Central Maximum Latency: 10 ms (0x000a)
        Number of CIS: 2
        CIS ID: 0x00
        Central to Peripheral Maximum SDU Size: 40
        Peripheral to Central Maximum SDU Size: 40
        Central to Peripheral PHY: LE 2M (0x02)
        Peripheral to Central PHY: LE 2M (0x02)
        Central to Peripheral Retransmission attempts: 0x02
        Peripheral to Central Retransmission attempts: 0x02
        CIS ID: 0x01
        Central to Peripheral Maximum SDU Size: 40
        Peripheral to Central Maximum SDU Size: 40
        Central to Peripheral PHY: LE 2M (0x02)
        Peripheral to Central PHY: LE 2M (0x02)
        Central to Peripheral Retransmission attempts: 0x02
        Peripheral to Central Retransmission attempts: 0x02
> HCI Event: Command Complete (0x0e) plen 10                                                                                                #2699 82.470480
      LE Set Connected Isochronous Group Parameters (0x08|0x0062) ncmd 1
        Status: Success (0x00)
        CIG ID: 0x00
        Number of Handles: 2
        Connection Handle #0: 7
        Connection Handle #1: 6
```
and the first CIS is connected as 
```
< HCI Command: LE Create Connected Isochronous Stream (0x08|0x0064) plen 5                                                                  #2826 83.089953
        Number of CIS: 1
        CIS Handle: 7
        ACL Handle: 0
```

and the second CIS as 
```
< HCI Command: LE Create Connected Isochronous Stream (0x08|0x0064) plen 5                                                                  #2850 83.541689
        Number of CIS: 1
        CIS Handle: 6
        ACL Handle: 1
```

Which arguably should work, but the difference in order of handles may indicate what the issue may be. 

**Environment (please complete the following information):**
 - Commit SHA or Version used: 97935bb0dcc621b6bcc2e65b5365b1f93c5993b4

**Additional context**
N/A

Thalley, 2023-06-06:
@cvinayak @mtpr-ot if this could be resolved before UPF next week it would be amazing, but it's also understandable if it cannot

Thalley, 2023-06-08:
> > @cvinayak @mtpr-ot if this could be resolved before UPF next week it would be amazing, but it's also understandable if it cannot
> 
> That is very strange, but I see Vinayak has a seemingly unrelated PR which would fix this? Let me know if we need to look further into this, beyond Vinayak's PR.

@mtpr-ot The PR from @cvinayak fixes the issue with the invalid parameters. It does, however, not fix the "issue" that the handles are switched after the first CIG. So the first CIG is still created and returns handles {6, 7}, but any subsequent CIGs are created with handles {7, 6}. The host of course does not care about this, but it's still odd that it's not returning the same thing each time :)

cvinayak, 2023-06-08:
> not fix the "issue" that the handles are switched after the first CIG. So the first CIG is still created and returns handles {6, 7}, but any subsequent CIGs are created with handles {7, 6}. The host [...]

The implementation uses memory pool, and hence the handles allocated will be LIFO based on the implementation of the memory pool. This behavior is as expected from the design, and is similar for ACL connection handles too. 

From the BT Spec., it is a `may` when it comes to re-use of handles, there is no recommendation about the values of the handles, except being from different number space depending on being advertising handles, connection handles etc...

> After a device deletes an ID or Handle (e.g. after the Controller deletes a
Connection_Handle for a disconnected ACL connection), it may reuse the ID or
Handle.