zephyrproject-rtos/zephyr #58901: GSOC 2023 Project - Milestone 1

Emna-Rekik, 2023-06-06:
# Test Plan for GSOC 2023 Project - Milestone 1

This issue serves as a successor to the enhancement issue that outlines the comprehensive milestones and deliverables of the entire project.

| Issue ID | Title |
|----------|-------------|
| #58495   | net: http: gsoc 2023 project outline |

## Overview of Test Plan: Testing the Prototype of an HTTP/2.0 Server (Linux)

The purpose of this test plan is to outline the requirements and features to test for implementing an HTTP/2 server, such as establishing a TCP connection, performing TLS handshake, handling connection preface, supporting multiplexing, stream prioritizing, header compression (HPACK), flow control, and server push.

## Testing Strategy

### 1. Testing Approach

The testing approach will involve a combination of unit testing, integration testing, and system testing:

1. **Unit Testing**:
   - Execute test suites to ensure proper error handling and resilience against abnormal situations.

2. **Integration Testing**:
   - Focus on testing the interactions between different components of the server, including the TCP connection establishment, TLS handshake, connection preface handling, stream communication, and other features.

3. **System Testing**:
   - Conduct end-to-end tests to verify the overall functionality and behavior of the HTTP/2 server.

### 2. Performance Testing

Performance testing aims to measure and evaluate the performance characteristics of the HTTP/2 server under various scenarios:

- Measure response time, resource utilization, and other relevant metrics to assess server performance.
- Validate the server's ability to handle a various number of concurrent connections and streams efficiently.

### 3. Security Testing

Security testing ensures the server's robustness and secure handling of connections:

- Perform security testing to identify and address vulnerabilities in the HTTP/2 server implementation.
- Conduct tests to ensure secure handling of TLS connections.

## Environment

- Operating System: Ubuntu machine
- Programming Language: C language
- APIs: Posix APIs

## Requirements to be Tested

1. **Establishing a TCP Connection**:
   - Verify that the server can listen on a specific port and accept incoming TCP connections.
   - Test the handling of multiple concurrent connections.

2. **Performing TLS Handshake**:
   - Test the server's ability to negotiate a TLS handshake with the client.
   - Validate the server's handling of various TLS certificates, including self-signed certificates.

3. **Connection Preface**:
   - Ensure that the server can receive a connection preface from the client and respond with its own connection preface.
   - Validate the handling of errors or invalid connection preface.

4. **Starting Stream Communication**:
   - Test the server's ability to initiate and handle stream communication after the TCP connection and TLS handshake.
   - Verify that the server can receive and process HTTP/2 frames and handle data transmission over the established streams.
   - Validate the handling of stream-specific headers, data frames, and any other relevant HTTP/2 protocol elements.
   - Test scenarios involving multiple streams running concurrently and ensure proper handling and synchronization.
   - Validate the server's behavior in cases of abnormal or unexpected stream communication termination.

## Features to be Tested (To Enhance Performance):

1. **Multiplexing**
2. **Stream Prioritizing**
3. **Header Compression (HPACK)**
4. **Flow Control**
5. **Server Push**

## Risk Assessment:
Server push in the HTTP/2 protocol presents potential risks and considerations, especially when targeting embedded systems. The proactive nature of server push, where the server can send objects to the client without explicit client requests, introduces the possibility of performance issues. Some of the associated risks include:

- **Performance Impact**: Pushing excessive or unnecessary objects to clients can negatively impact server and network performance, leading to decreased responsiveness and increased latency.

- **Resource Utilization**: Inefficient handling of resource allocation can result in excessive resource usage, potentially affecting the overall server performance.

## Task Checklists

- [ ] Establishing a TCP Connection.
- [ ] Performing TLS Handshake.
- [ ] Connection Preface.
- [ ] Starting Stream Communication.

cfriedt, 2023-06-08:
We might be able to assign tasks to Emna once she has made a commit. I'm not sure.

cfriedt, 2023-06-13:
@Emna-Rekik, @rlubos - I wonder if the format of this should be more like an RFC. What do you think? It can be refined and then probably presented at the Networking Forum at some point.

rlubos, 2023-06-14:
> It can be refined and then probably presented at the Networking Forum for feedback / discussion. IIRC, it's once a month, but I'm not sure of the next meeting.

Next Networking Forum is due on 4th of July, not sure about the attendance though given this specific date. But for sure we could put this on the agenda.

cfriedt, 2023-06-23:
@Emna-Rekik - could you possibly cut & paste the test plan portion of this issue somewhere else? Let's try to keep this issue description as more or less a checklist of subtasks

clamattia, 2024-03-25:
Hello Zephyr Community

We are very interested in this feature and are curious, whether there is still work in progress on this.
Is the `http server` on the roadmap/planned?

rlubos, 2024-03-25:
@clamattia Yes, we're working on wrapping the project up and integrating it with Zephyr (hopefully for 3.7 release).

clamattia, 2024-03-25:
> @clamattia Yes, we're working on wrapping the project up and integrating it with Zephyr (hopefully for 3.7 release).

Cool, thanks.

closed 2024-08-25 by cfriedt